christianhaitian/arkos #693: Please add .cmd file extension to PC98 es_systems.cfg entry

zomgugoff, 2023-04-16:
### Before submitting your issue or feature request.  DO NOT REQUEST ASSISTANTS FOR PRELOADED IMAGES.  I do not support or condone the use of preloaded ArkOS images.  Preloaded meaning images that come with copyrighted roms preincluded.  Look for support from the creator or supplier of those preloaded images, not from here!  If a ticket is submitted for assistance for such images or if there's an indiciation that a preloaded image is what the support request is for, the issue ticket will be closed with no further response.

1. ***This is not Anbernic Stock Emuelec or 351Elec.***  ArkOS is a different distribution from those systems and will have differing features.  The only somewhat similarity is that they both use EmulationStation as a frontend.  Even with that, the versions of EmulationStation that's used in between ArkOS and Anbernic Emuelec/351Elec is different with different capabilities.  If there are features you prefer with Anbernic Emuelec/351Elec, just use those distributions instead.  The choice is yours.

2. Have you reviewed the Frequently Asked Questions available at https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---RG351P ?
Yes

3. If your issue is related to being able to run certain emulators or ports, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?
Yes

4. Have you reviewed the ***known issues*** section available at https://github.com/christianhaitian/arkos/wiki/Known-issues ?
Yes

### Describe the Issue (If applicable)
A clear and concise description of what the bug is.
Feature request: Please add .cmd file extension to PC98 es_systems.cfg entry. Nekop2 and NP2Kai support .cmd files for mounting multiple disk images on boot. The file format is:

np2kai "image file for FDD1" "image file for FDD2"

The image files can be any support disk image file, so it can mount hard drive images as well as CD images.

The 'np2kai' command at the beginning of the line is necessary and works with both emulators. The double quotes around the file names may not be necessary, but handles spaces in filenames.

### How can the issue be reproduced? (If applicable)

### What device are you using?
RG353M

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen
2.0 (04012023)

### Anything else you'd like to include that may help to review this issue or feature request?
example file contents:
np2kai "Mugen Senshi Valis 2 (1986)(Nihon Telenet)(Disc 1 of 3)(Disk A).FDI" "Mugen Senshi Valis 2 (1986)(Nihon Telenet)(Disc 2 of 3)(Disk B).FDI"

christianhaitian, 2023-04-16:
Sure

christianhaitian, 2023-04-29:
This has been accommodated with the latest update.

closed 2023-04-29 by christianhaitian