christianhaitian/arkos #694: RG353V & Launch image on 2 SD set up

nithou, 2023-04-18:
### Describe the Issue (If applicable)

I recently switched to a 2 SD set up. One 32 GB SD for the system (on which I installed a clean version of ArkOS), reimported my settings through the options menu, and a second sd (for the games) that I formated too and went to the Options system to proceed to the 2 SD Set up. 

Since then however, I can't make any launchimage or ASCII to work, the screen goes black then flashs dark grey but doesn't load any image. I tried with the original images in bpth SD1 EASYROM partitions and SD2 launchimages folders but none seem to work.

The system seem to display a brief error but I can't spot what it's written and can't find anything log related 

### What device are you using?

RG353V v2 screen

### What version of ArkOS are you on?
2.0 (04012023)

christianhaitian, 2023-04-18:
Yes this is an issue with the v2 screen on ArkOS that was just recently resolved.  Try the attached file.  Just take the dtb in it and rename it to rk3566-OC.dtb and copy it to the BOOT partition overwriting the existing file there.  You can do this from your computer with a card reader/writer since the BOOT partition is fat32 and can be modified from Windows and Mac.  Once completed, just reinsert into your unit and the splash screen should work. 

[rk3566-353v.zip](https://github.com/christianhaitian/arkos/files/11259078/rk3566-353v.zip)

nithou, 2023-04-18:
Thanks, works perfectly with the file! It takes the launchimages from the rom partition of SD1 :)

christianhaitian, 2023-04-18:
Excellent.  Take care

closed 2023-04-18 by christianhaitian