christianhaitian/arkos #696: After OTA Update, Options gone

lauren7ino, 2023-04-19:
Greetings. Sorry my basic English.
Seeing if anyone can confirm this problem...
RG353M, installed ArkOS_RG353M_v2.0_03302023.img, configured Wifi, run Update, confirmed with "OK", then RetroArch disappeared from main screen, and Option menu options mostly disappeared, only remaining *Advanced + Bluetooth + Update + Wifi.
This is the second time this happens to me.

christianhaitian, 2023-04-19:
Don't use "Parse Gamelists only".  It doesn't save much in initial loading time anyway.

closed 2023-04-19 by christianhaitian

lauren7ino, 2023-04-20:
First of all, thank you for your assistance.

I’m only alerting for the consequences of losing all those settings.
1) An alert should be issued,
2) It should not happen (?) since gamelists are for games (as the name implies) and it took me hours to figure out why the settings were missing (I reinstalled ArkOS from scratch 4x) 
3) I’m not alone. Found other people online complaining about the same missing settings with no answer.

Thanks and keep up the good work!

—Laurentino Martins

lauren7ino, 2023-04-29:
Thank you for your help and your excellent work. I've donated for the project.

Related to the above question, in Amiga, I've created .m3u files for all multi-disk games, since PUAE uses the .m3u to auto-install the disks in multiple drives. It works fine.

Now the annoyance: The game launcher (EmulationStation?) lists these games twice.
Note: gameslist.xml only lists the .zip file.

Without the "Parse Gamelists only" problematic setting, is there a way to ignore the .m3u files?

Thanks in advance

christianhaitian, 2023-04-29:
You can manually edit the es_systems.cfg file located in `/etc/emulationstation`.  Just remove the extensions for Amiga you do not wish to show.  Make sure you use a unix/linux compatible emulator.  Not the default notepad or wordpad editor in Windows since that will add additional characters that linux/unix don't like.

lauren7ino, 2023-04-30:
Sorry for the delays in the answer, I've been quite busy.
Finally I installed Ubuntu in my Parallels Mac mini and managed to edit the file you mentioned.
I removed both the .m3u and .M3U references in Amiga and the duplications disappeared.
Thank you very much!

christianhaitian, 2023-05-01:
Awesome.  You could've also just used a ssh terminal program to remote into ArkOS while powered on to make that change as well.

lauren7ino, 2023-05-03:
(Do you have a personal email address I could use?)

You may want to know you are distributing ArkOS with a pre-configured item. 
Last time I installed ArkOS from scratch I found a WIFI Network or a Bluetooth device pre-configred.
I'm not sure which, it was some time ago.
You may want to remove this in the next release.

Thanks, and keep up the good work.
Sorry my basic English.

christianhaitian, 2023-05-03:
Thanks for reporting this.  I'm sure it wasn't wifi.  The one preconfigured item in the wifi menu is for ethernet based connections which is needed if you ever plan to connected a hardwired ethernet dongle.  I'll double check that bluetooth doesn't have a preconfigured device in there.

lauren7ino, 2023-05-03:
I have an RG353M with integrated WiFi, so this is why I never understood what that WiFi connection was for in the WiFi menu.
Thanks

Another question, regarding the access with SSH, on the FAQ you talk about "Go to the Options menu and select NETWORK INFO so you can see your the assigned IP to your device.".
I must be blind. I can't find this "NETWORK INFO" menu?

christianhaitian, 2023-05-03:
I need to update the faq.  I removed that as you can just hit start and scroll to the bottom of the screen to see your ip info.

lauren7ino, 2023-05-04:
Ah, that explains it :)

Sorry, I hope I'm not abusing your assistance, but I have one final question:
Do you have any ideia why the port Cannonball (Out Run) doesn't save or can't load (?) the user settings? I read somewhere that might have to do with permissions.
Thanks

christianhaitian, 2023-05-04:
What format is your roms2 card in?

lauren7ino, 2023-05-05:
ExFAT

Do you know anyone that is able to save the settings?

I've searched all around the partitions and failed to find a configuration file (although I wasn't sure what I was looking for)

christianhaitian, 2023-05-05:
We'll have to test and see what's going on with this.