christianhaitian/arkos #699: RG351V can't find the RK3566-OC.dtb to delete after first boot.

rafhaelmallorga, 2023-04-22:
### Describe the Issue (If applicable)
I'm trying to install ArcOS on my new ABERNIC RG351V, but I had the same problem reported on [Issue 620](https://github.com/christianhaitian/arkos/issues/620).

So I followed the tips but I can't open the files from the MicroSD to delete and replace RK3566-OC.dtb > RK3566-OC.dtb.v2.

I tried to read the SD on Mac and on Windows.

- On Mac it read the SD but I cant access the data inside to replace the files, it didn't mount. It shows a message saying that it's not readable by computer, so I can't open with finder.

[image]

[image]

I follow the tip of @scspijker to check if the partition is on ***Microsoft Basic Data*** and it's in the right format:

[image]

- On Windows it don't open too, but I can check the partitions with a software and they are there, but I can't access with file manager too.

Someone can help with this?

### How can the issue be reproduced? (If applicable)
Install the ArcOS in a new device.

### What device are you using?
**ABERNIC RG351V**

### What version of ArkOS are you on?
**V2.0**

### Anything else you'd like to include that may help to review this issue or feature request?

RetroGFX, 2023-04-22:
The partition the dtb is on its Windows readable..u shouldn't need extra software to see if bud

rafhaelmallorga, 2023-04-22:
Yes, but it is not reading

[image]

Using ***MiniTool Partition***

[image]

RetroGFX, 2023-04-22:
U need to go into disk manager and assign a drive letter

rafhaelmallorga, 2023-04-22:
It don't allow me to change the drive letter.

[image]

By CMD it don't list the MicroSD

[image]

RetroGFX, 2023-04-22:
I don't know what that says I don't speak Spanish, but u don't have a drive letter on any of those partitions for the sd.. That's the issue

RetroGFX, 2023-04-22:
Your other option is take the sd out, reboot the PC, and try to reflash and see if it works right.. Use win32diskimager..I don't recommend Rufus or etcher

christianhaitian, 2023-04-22:
It looks like you're on Windows 10 so you definitely shouldn't be having this issue.  First off, you're referencing an issue related to the RG353V and not the RG351V.  The RG351V fix you would need is [here](https://github.com/christianhaitian/arkos/issues/662#issuecomment-1511860062).  Looking at the the images you posted, looks like the processes started the exfat partition creation and expansion as you have an exfat partition already which shouldn't already exist if this is from a brand new image placed on the sd card.  So that setup maybe corrupted.  Please try the following:

1.  Apply a fresh image on the SD card.  Make sure it's one that is specific for your unit.  There's a specific RG351V image.  The RG353V image will not work on the RG351V as they have 2 different chipsets.
2. Make the changes per the link I posted above.
3. Then insert the sd card into your unit and see if that works.

RetroGFX, 2023-04-22:
Lol..I totally missed that @christianhaitian 🤣😂

rafhaelmallorga, 2023-04-22:
I`m using the right image file from RG351V. 

I just formated again and reinstall. After that I boot on the device, wait the 2 boots, until start play the background music. So I think that it worked.

But when I try to read the SD on windows it dont allow me.

I tryied to read on Linux, and in this path I found this files, maybe it give you some clue:
[image]

[image]

The boot file is empity.

rafhaelmallorga, 2023-04-23:
I got it.

I was using the rufus to write the image. And for some reason it wasnt generating the BOOT folder.

I used ***win32diskimage*** as told by @RetroGFX and the ROOT file appeard.

So I just replace ***rk3326-rg351v-linux.dtb*** and boot. 

It worked perfectly.

closed 2023-04-23 by rafhaelmallorga

christianhaitian, 2023-04-23:
Interesting.  So Rufus affects that it seems.  Good to know.

RetroGFX, 2023-04-23:
> Interesting. So Rufus affects that it seems. Good to know.

Yea I find Rufus and etcher both to be equally buggy it seems..I prefer simple flash tools myself

rafhaelmallorga, 2023-04-23:
Yes, but I'm still having problem to reading the partition. I'm trying to configure the second card now, and following the tutorial I had to copy de bios folder from SD1 to SD2.

But I can't access that yet, I'll try it agaí d on Mac and Linux.