christianhaitian/arkos #701: m3u files for Amiga/PUAE

lauren7ino, 2023-04-29:
Hello,
PUAE Amiga emulator uses m3u files for all multi-disk games to auto-install the disks in multiple Amiga drives. It works perfectly in all games that support multi-drives (most do).
Suggestion: Please expand the existing PS1 "m3u" settings to the Amiga also (delete/generate/show)
Thanks

lauren7ino, 2023-04-29:
Addendum:
I've hand generated .m3u files for all (.zip) amiga games that contain more than 1 file, and works fine on PUAE.

Problem:
My tests seem to show that Amiga/PUAE can't launch .m3u files, so to avoid duplications in the games list, .m3u extension must be removed from es_systems.cfg (thanks for the tip)

christianhaitian, 2023-04-30:
So puae doesn’t support m3u files?  So this request should be closed?

lauren7ino, 2023-04-30:
PUAE uses m3u in game to auto install/swap Amiga game disks.

I suggested hiding m3u files from Emulation Station in the Amiga list, because:
1) Avoids games duplications.
2) Launching m3u with PUAE as default emulator don't seem to work (on my setup).

christianhaitian, 2023-04-30:
Does it work with uae4arm?

lauren7ino, 2023-04-30:
Just to clarify, one example: Gods (game)
- Gods.zip contains Gods_Disk1.adf + Gods_Disk2.adf
- Gods.m3u contains Gods_Disk1.adf + Gods_Disk2.adf
- gameslist.xml contains Gods.zip

EmulationStation lists both the Gods.m3u + Gods.zip

Launching Gods.m3u doesn't work neither with Uae4arm nor with PUAE as the default emulators.
Launching Gods.zip works in both.

christianhaitian, 2023-05-05:
Accommodated with today's update.

closed 2023-05-05 by christianhaitian

lauren7ino, 2023-05-05:
Updated. I see no difference in the Configuration menus.
What am I missing?
Thanks

christianhaitian, 2023-05-05:
.m3u extensions have been removed in the es_systems.cfg file for Amiga and Amiga CD32