christianhaitian/arkos #702: Failed to run Love2d (.love) files

drokrypto, 2023-04-30:
Whenever I try to run a .love file I receive an error:

[image]

`Error

[love "boot.lua"]:321: No code to run
Your game might be packaged incorrectly.
Make sure main.lua its at the top level of the zip.

Traceback

[love "callback.lua"]:228: in function 'handler'
[C]: in function 'error'
[C]: in function 'xpcall'
[C]: in function 'xpcall'

Press Ctrl+C or tap to copy this error`

I have made sure to package the game correctly. The `main.love` file is at the root of the .zip/.love file.

[image]

I have used my own files as well the the test file below:
https://github.com/christianhaitian/arkos/issues/294#issuecomment-1133845765

**ArkOS version:** 2.0 (04272023)
**Device:** RG353M
**Host OS:** macOS Ventura (13.3.1)

christianhaitian, 2023-04-30:
Is it a game that’s compatible with version 11.4 of Love2d? https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information#love2d

drokrypto, 2023-05-02:
Yes, and the test referenced in the bug report does not work either, which is also compatible with version 11.4.

christianhaitian, 2023-05-02:
Can you post your love package here so I can review?

drokrypto, 2023-05-02:
> Can you post your love package here so I can review?

Sure, the following is the same test from [the comment](https://github.com/christianhaitian/arkos/issues/294#issuecomment-1133845765)  that I linked in the bug report:

[arktest.zip](https://github.com/christianhaitian/arkos/files/8749169/arktest.zip)

christianhaitian, 2023-06-24:
Not yet.  sorry I haven't looked at this yet.

FlyingDojo, 2024-03-17:
@drokrypto Is your .love file laid out as below?

(roms folder)
-arktest.zip(renamed to "arktest.love")
--main.lua
--conf.lua
--xa-final.xm
--example-folder
---example-folder-script.lua

I've tried your file and it works completely fine for me as laid out above.

christianhaitian, 2024-10-07:
Closing this issue as there's another similar issue ([1180](https://github.com/christianhaitian/arkos/issues/1180)) posted with some suggestions for remediation.