christianhaitian/arkos #703: Search button (X) not working in Retroarch in RG353M

droople, 2023-05-01:
### Describe the Issue (If applicable)
A clear and concise description of what the bug is.
The search button (the X button by default) not working in Retroarch in RG353M

### How can the issue be reproduced? (If applicable)

Enter the Retroarch, press X button, the search UI doesn't pop up

### What device are you using?
Rg353m

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen

ArkOs 2.0 04012023

### Anything else you'd like to include that may help to review this issue or feature request?
Someone else has the same issue on rg351 with Amberelec

christianhaitian, 2023-05-01:
Hi,

Yes, we do that on purpose because when it's triggered you exit from retroarch with it still active, it disables the ability to use most hotkeys in retroarch which then causes confusion for most folks on how to properly exit retroarch based emulation.  More information about this issue is available [here](https://forums.libretro.com/t/rare-hotkey-issue-on-windows-10-retroarch-pc/21311/3).
The patch applied in ArkOS is this <https://github.com/christianhaitian/rk3566_core_builds/blob/master/patches/retroarch-patch-0004-disable_search.patch>. 
The one that AmberElec applies is this <https://github.com/AmberELEC/AmberELEC/blob/dev/packages/games/tools/retroarch/patches/retroarch-02_disable_search.patch>.

For ArkOS, if you wish to have that feature enabled, you'll have to build retroarch yourself and disable that patch.

closed 2023-05-01 by christianhaitian

droople, 2023-05-03:
Thank you very much.