christianhaitian/arkos #704: overclocking

westhecool, 2023-05-01:
I was wondering if it would be possible to implement overclocking. I understand there is another discussion about this and that it would not gain a lot of performance. But from what I understand this is just because of heat dissipation? And I was thinking of modding my RGB10 max 2 to maybe install it a heat sink or so.

christianhaitian, 2023-05-01:
Sorry, ArkOS doesn’t support the rgb10max.  Just the 3.5” rgb10.  Last time overclocking was tested on the regular rgb10, performance was worse.  Not sure if it would be better on the rgb10max.  There you’d have to test on retrooz.

closed 2023-05-01 by christianhaitian

westhecool, 2023-05-01:
> Sorry, ArkOS doesn’t support the rgb10max. Just the 3.5” rgb10. Last time overclocking was tested on the regular rgb10, performance was worse. Not sure if it would be better on the rgb10max. There you’d have to test on retrooz.

Oh I thought it supported that console as well

christianhaitian, 2023-05-01:
Nope.  Just RetroOZ has official support for the rgb10max units.  Maybe UnofficialOS may support it down the line as well.