christianhaitian/arkos #705: Dreamcast Games Not Found

kielyallen, 2023-05-01:
Following the [Wiki](https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---RG353M) and [Emulator Information](https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information#dreamcast),  after adding Dreamcast bios and roms to all possible directories, multiple game formats, Dreamcast games were not found by emulation station on a fresh installation of ArkOS 04272023. After searching emulationstation posts for a similar issue, I noticed `/home/ark/.emulationstation/es_systems.cfg` is not found, which should be loaded first before then checking for the file in `/etc/emulationstation/es_systems.cfg`.  The latter exists, and when opened contains the appropriate XML for the Dreamcast, but with one issue.

```
        <system>
                <name>dreamcast</name>
                <fullname>Sega - Dreamcast</fullname>
                <path>/roms2/dreamcast/</path>
                <extension>.7z .7Z .gdi .GDI .cdi .CDI .cue .CUE .chd .CHD .m3u .M3U</extension>
                <command>sudo perfmax %EMULATOR% %CORE%; nice -n -19 /usr/local/bin/dreamcast.sh %EMULATOR% %CORE% %ROM%; sudo perfnorm</command>
                   <emulators>
                      <emulator name="retroarch32">
                        <cores>
                          <core>flycast32_rumble</core>
                          <core>flycast_xtreme</core>
                          <core>reicast_xtreme</core>
                        </cores>
                      </emulator>
                      <emulator name="retroarch">
                        <cores>
                          <core>flycast_rumble</core>
                          <core>flycast</core>
                        </cores>
                      </emulator>
                   </emulators>
                <platform>dreamcast</platform>
                <theme>dreamcast</theme>
        </system>
```

The problem is `<path>/roms2/dreamcast/</path>`. `/roms2` is not mounted, but exists in `/`. `/roms` is mounted.

`/dev/mmcblk1p5 /roms exfat rw,noatime,uid=1002,gid=1002,fmask=0000,dmask=0000,allow_utime=0022,iocharset=utf8,errors=remount-ro 0 0`

Once the correct path is changed to `<path>/roms/dreamcast/</path>`, ArkOS sees the roms. This is also the case for some other systems like Amiga.

Is it possible that some [sed commands](https://github.com/christianhaitian/arkos/blob/main/Update-RG353M.sh#L1507) were not run in update?

Able to provide further logs on request if needed.

ArkOS version: 2.0 (04272023)
Device: RG353M
HostOS: Windows 10

christianhaitian, 2023-05-01:
Interesting.  Did you have a 2 sd card setup at one point in time?  That command you point to in the update script should not have ran unless roms2 was found in the `/etc/fstab` file which is only put in there after doing a switch to sd2 setup.

kielyallen, 2023-05-01:
This was done with a clean 256GB microsd, writing the 03302023 .img directly with rufus and installing from the INT slot on the RG353M. I then updated via WiFi to 04272023. There is no second card inserted and there never has been.

christianhaitian, 2023-05-05:
Addressed with today's update.

closed 2023-05-05 by christianhaitian