christianhaitian/arkos #706: rg353M doesn't sleep

eteream, 2023-05-05:
After pushing the power button the screen turns off, the led remains green (or red) and the battery drains more than usual.

On stock firmware the led goes off. I seems that the led is linked by hardware (not software) to the power state, I have read, not 100% sure.

christianhaitian, 2023-05-05:
https://github.com/christianhaitian/arkos/issues/643#issuecomment-1434396423

closed 2023-05-05 by christianhaitian

eteream, 2023-05-05:
353 devices crash only when playing a video when the power button is pushed, 100% sure, never under any other circumstances I have seen a crash with the anbernic firmware. All you have to do is to be sure that no video is played when going to sleep.

I have read that has to do something to do with the audio buffer not recovering after sleep, not tested myself.

christianhaitian, 2023-05-05:
If refering to a video preview under emulationstation, ArkOS has workaround for that issue.  Others have reported issues with sleep working reliably even while sleeping in game or in the ES menu without video previews working and even on stock OS.  It seems like it's not a consistent issue as my rg353m deep sleeps fine.  If you follow the instructions, in the comment I posted above, you can reenable deep sleep.

eteream, 2023-05-06:
After enabling Deep Sleep on "Options/Advanced/Sleep - Switch to Deep Sleep Support" the green led still doesn't turn off, I guess that means it's not really on deep sleep.

christianhaitian, 2023-05-06:
the led being on doesn't mean it's not in deep sleep.  the led doesn't impact the battery life.  it's just a means to understand that the unit is on.  led's don't burn much battery at all.

eteream, 2023-05-07:
From 99% to 96% in battery in 4 hours. So it works!

Thanks for the help Christian, much appreciated.