christianhaitian/arkos #711: Theme Master

lauren7ino, 2023-05-10:
I believe it was the intent of ArkOS to install the Theme Master the same way Port Master is installed on a secondary empty card. Unfortunately ThemeMaster directory + content is missing, requiring manual installation.

christianhaitian, 2023-05-10:
Which unit and what date of the image are you using?  I updated the images for the rk3566 devices as of 5/4/2023 that should have resolved this.

lauren7ino, 2023-05-10:
I’m not sure how to see the version, but I have no pending updates according to ArkOS.
I don’t know what to say. I find strange that /Roma/tools contains no ThemeMaster directory but contains the PortMaster directory.
Anbernic RG353M

christianhaitian, 2023-05-10:
When did you download and load the image?

lauren7ino, 2023-05-10:
OK, I managed to capture the version displayed on the boot:  ArkOS 2.0 (05042023)

I can attest that ArkOS\ArkOS_RG353M_v2.0_05042023.img\3.Basic data partition.img\roms\tools\ThemeMaster\ exists, so the only conclusion is that besides ArkOS is up to date, the base image I used was older, probably ArkOS_RG353M_v2.0_03302023.img

christianhaitian, 2023-05-10:
ok.  that makes sense since that version did have that issue.  Also, you can see the current version you have of ArkOS from emulationstation by pressing the start button and scrolling to the bottom of the screen.  Thanks and enjoy

lauren7ino, 2023-05-10:
OK thanks. Mystery solved.
So, the OTA updates in ArkOS only update from IMG 05-04-2023 forward?

christianhaitian, 2023-05-10:
correct.