christianhaitian/arkos #713: Possible performance-related bug

lodx-xd, 2023-05-12:
Device: RK2023
ArkOS version: v2.0 (05/04/2023)

I'm not sure exactly if this is a bug or maybe a power saving feature, but recently I noticed that I was having slowdowns (although the fps counter didn't register any drops) when using the gtuv50 crt shader on systems that use the "retroarch" frontend, snes, genesis and neogeo were the systems I tested, I put this to bed because I thought it was a matter of this shader being too heavy for the RK3566.

But as soon as I boot a game that uses an external emulator or the "retroarch32" version and go back to the previous mentioned systems in "retroarch" using the same gtuv50 shader, this time I have the "normal" performance expected from these games without any slowdown, and the performance remains constant until the reboot/shutdown of the device, and then again the "performance problem" reappears needing to redo this "workaround".

What makes me imagine (based on my ignorance alone about the subject), that the performance of the device is being limited while heavier systems are not used, probably for power saving? I imagine that this is also the reason why we have two different versions of "retroarch" on ArkOS and not only for compatibility?

But having that said, I'm not sure if this slowdowns is expected on the "retroarch" and I should be using the "retroarch32" in fact if I want to use the more demanding shaders or if it is really a bug, if so sorry for take your time with this issue.

Unfortunately I could not yet test this issue in the v2.0 (05/11/2023) released a few hours ago because I'm still waiting for my wifi adapter to arrive by mail.

christianhaitian, 2023-05-12:
Sounds like a possible cpu governor related issue.  Do you notice this performance issue when no shader is used at all with retroarch?  Is this performance issue with just that specific shader?  Have you tried any other shader in retroarch and noticed a similar behavior?

lodx-xd, 2023-05-12:
When shaders are not used I couldn't notice any differences, but I believe that because the systems and games I tried are easy to run using an RK3566 even with this possible "issue", I tried to run something heavier (Bloody Roar II) using DuckStation and SwanStation present in the "retroarch" (not "retroarch32"), neither of them reached full speed the whole time, but it seemed to me that after the "workaround" that I mentioned earlier, it had less slowdown, but it is too inconsistent for me to confirm if is true or just a placebo effect in this case.

But with shaders it is easier to replicate the problem on less demanding systems, using other heavier shaders like crt-hyllian the slowness also appears, and after the "workaround" the same disappears.

christianhaitian, 2023-05-18:
Please see if this performance issue has been resolved with today's update.  Be sure a timezone is set in the os and ensure that power saver mode is disabled as it may cause performance issues in the OS as well.

lodx-xd, 2023-05-18:
I have updated to the latest version released a few hours ago.
I set my local time zone.
And make sure that the power save mode is disabled.

But the problem still remains, but only when I try to launch a game from the EmulationStation interface.

Please bear with me and my poor English skills.

I had noticed today with some tests before the update, that when I tried to do the aforementioned workaround (launch some game that uses "retroarch32") directly from the RetroArch menu in the EmulationStation (that is, open the "retrorach32" frontend and select the game and core manually) the workaround would not work when I went back to the EmulationStation and tried to launch a game from there. It only worked if I did the workaround directly from the EmulationStation interface.

After the update, seeing that it didn't solve the problem for me, I tried to run the "retroarch" frontend directly from its own menu in the EmulationStation, and who would say, it run the game perfectly without any slowdown, but this doesn't translate when I try to run the same core, game and shader through the EmulationStation interface without the workaround, I still need to do the workaround in this case, even if the game as running fine trough the RetroArch menu.

So that makes me think that the problem is related to EmulationStation somehow?

It is worth mentioning that in one of my attempts to launch a game directly from the RetroArch menu it run as slow as if I had launched the game from the EmulationStation interface, but I was not very attentive at the time and maybe I selected some wrong option, since I repeated the test several times after and all the other times it worked perfectly.

christianhaitian, 2023-05-18:
Is Bluetooth on on your device?  If so, Try disabling Bluetooth (F+R3) and see if that changes anything with the performance.

christianhaitian, 2023-05-18:
nevermind.  I forgot this is the rk2023 device.  Do me a favor and go to Options/Advanced/Sleep - Enable Deep Sleep.  See if there's any performance differences after enabling deep sleep.

lodx-xd, 2023-05-18:
I did the experiment, but no avail, performance remained slow, with deep sleep enable or not.

About this: 

> It is worth mentioning that in one of my attempts to launch a game directly from the RetroArch menu it run as slow as if I had launched the game from the EmulationStation interface, but I was not [...]

I was able to replicate this situation, in the case every time I launch a game using the "retroarch" frontend from the EmulationStation interface, without the workaround, it also causes the retroarch launched from the RetroArch menu to run with the same slow performance, only returning to "normal" if I reset the device or do the workaround.

christianhaitian, 2023-05-18:
What is the make and model of your sd card?  Have you tried the Stock OS to see if the issue occurs there?

lodx-xd, 2023-05-19:
A 64 GB Samsung Evo Plus and a 64GB SanDisk Ultra, already try to swap the cards, or run the games from only on of the sd cards, for no avail, if this information is relevant.

Using JELOS I have no performance problems, the game will launch at normal performance using the same game, core and shader.

christianhaitian, 2023-05-19:
Please post the link to the image you downloaded to load on to the sd card.

christianhaitian, 2023-05-19:
I'm trying to replicate this issue on my end and can't yet.

lauren7ino, 2023-05-19:
@lodx-xd
In case I find the time to replicate your experience in my RG353M, in JELOS which core, shader and game are you using for the tests?
Thanks

lodx-xd, 2023-05-19:
@christianhaitian 
> Please post the link to the image you downloaded to load on to the sd card.

The ArkOS (Updated 05/04/2023) I got it  from the wiki page from the dgrive link https://drive.google.com/file/d/11IYLZQVUA3QpNJs6-oj4ep7lnAq9skvU/view after that I just download the latest updates through WiFi.

@lauren7ino 
Using the current Snes9x core, shader is the crt gtuv50 and the games I'm trying while testing this issue here are Mega Man X3 and Terranigma from snes and Mega Man: The Wily Wars from the Genesis/Mega Drive, but everything from these two system I get slowdowns without booting a higher system before, just picked theses games to test because I'm familiar with the performance they have.

lauren7ino, 2023-05-19:
Try going to RetroArch's main menu, select 'Online Updater', and then select 'Update Cores'?
Do the same on RetroArch 32.
Update everything.

christianhaitian, 2023-05-19:
Ok, I was able to recreate the issue with the parameters specified.  I can do some troubleshooting now.

christianhaitian, 2023-05-19:
Ok.  I think the issue is related to a bug in the governors.  Can you help me test and see if this change fixes the performance issues you're seeing?

1. Take the attached compressed file (arkosbackup.tar.gz) and put it in your roms/backup folder. Do not uncompress the contents! Just place the file as is in your roms/backup folder. If you don't have a backup folder in the roms folder, just create it.  If you're using a 2 SD card setup on your device, this must be done from SD1(TF1 Slot).

2. Then go into the Emulationstation Menu, scroll to Options, hit A, then go to Advanced, hit A, then scroll all the way down to Restore Settings, hit A.

3. Once completed, reboot your device and test if this issue is resolved without needing to do the workaround.

[arkosbackup.tar.gz](https://github.com/christianhaitian/arkos/files/11517219/arkosbackup.tar.gz)

This test should work on any of the rk3566 devices (RG353, RG503 and RK2023 units)

Thanks,

lodx-xd, 2023-05-19:
@christianhaitian 

Did the test and now no more issues as far I could test, amazing work and thanks for your time on this issue.

christianhaitian, 2023-05-19:
Great.  I'll include that fix in the next update.

christianhaitian, 2023-05-20:
I released a minor update for this fix.  Thanks again for reporting and confirming the fix worked.

closed 2023-05-20 by christianhaitian

lauren7ino, 2023-05-20:
Hi
I tried N64 Ridge Racer on Standalone Glide 64MK2 and it runs smoothly without hiccups. I believe that didn’t happen before.
Sorry, I can’t help any further (busy bee).
Thanks