christianhaitian/arkos #714: RG353V: Deep sleep issue (05112023 release)

MetroidFusion, 2023-05-12:
Hi

The deep sleep function on the RG353V seems broken in the 05112023 release. When pressing the power button - to wake the device from deep sleep - the green light indicator turns off, and the device becomes unresponsive. The only way to get it up and running again, is by hitting the reset button.

/M

christianhaitian, 2023-05-12:
yes, i'm aware.  The updated uboot breaks deep sleep for all units. So we have increased stability but no deep sleep support at the moment.  I'll have a workaround for the issue soon.  For now, light sleep works.

MetroidFusion, 2023-05-12:
Sounds good - thank you very much. I'm looking forward to the workaround for the issue :)

/M

christianhaitian, 2023-05-18:
This deep sleep issue should be resolved now with today's update.

closed 2023-05-18 by christianhaitian

JulyIghor, 2023-08-25:
> This deep sleep issue should be resolved now with today's update.

looks like the issue is back now

dsv9, 2024-01-16:
@christianhaitian I'm currently experiencing it on a RG353VS, with the latest build downloaded and flashed on the 14th of January.

christianhaitian, 2024-01-16:
Did you do the ota update and switch to deep sleep from options/advanced section?

christianhaitian, 2024-01-16:
The light will still be on during deep sleep.  That was done on purpose so you know the difference between when the device is asleep or fully off.  The LED drain is neglible as LEDs don't drain much from the battery at all.  Not even 1% over days of being on.

aenertia, 2024-07-24:
Using the latest update I enabled deep sleep and now my device won't boot at all with the arkos OS card installed. LED went red when restarting per prompt and never changed. Reset I would get amber/red and no boot. Ejecting the ArkOS tf I was able to boot into Rocknix (installed to emmc from a custom build from the 22/07/2024). What setting / file do I need to modify on what partition to revert the behavior. 

With rocknix on emmc I was able to boot into arkos when the tf was inserted without issue (and the reboot bug is not present when the Mainline uboot/SPL used in rocknix is on the emmc). 

Device: rg353M

aenertia, 2024-07-24:
Update - replicated on my rg353V