christianhaitian/arkos #715: Remove corrupt sram.dat from PX68K core on start/demand

zomgugoff, 2023-05-12:
### Before submitting your issue or feature request.  DO NOT REQUEST ASSISTANTS FOR PRELOADED IMAGES.  I do not support or condone the use of preloaded ArkOS images.  Preloaded meaning images that come with copyrighted roms preincluded.  Look for support from the creator or supplier of those preloaded images, not from here!  If a ticket is submitted for assistance for such images or if there's an indiciation that a preloaded image is what the support request is for, the issue ticket will be closed with no further response.

1. ***This is not the Stock OS that comes with your device, AmberElec, JelOS, or UnOfficialOS.***  ArkOS is a different distribution from those systems and will have differing features.  The only somewhat similarity is that they both use EmulationStation as a frontend.  Even with that, the versions of EmulationStation that's used in between ArkOS and these other distributions is different with different capabilities.  If there are features you prefer with the other available distributions, just use those distributions instead.  The choice is yours.

2. Have you reviewed the Frequently Asked Questions section available for your supported device in the wiki?
Yes

3. If your issue is related to being able to run certain emulators or ports, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?
Yes

4. Have you reviewed the ***known issues*** section available at https://github.com/christianhaitian/arkos/wiki/Known-issues ?
Yes

### Describe the Issue (If applicable)
A clear and concise description of what the bug is.
PX68K has an issue with the sram.dat (/roms[2]/bios/keropi/sram.dat) file that is generated/updated when the emulator starts/exits. It can become corrupt and will prevent software from booting in the emulator resulting in a black screen, sometimes with a single line of Japanese text in the middle of it. The easiest way to fix it is to remove the corrupt file.

An not-so-elegent fix is to simply remove it right before the emulator launches. Less so, an option in a menu somewhere to do it.

### How can the issue be reproduced? (If applicable)
It's sort of hit-and-miss as to if/when the corruption can occur. Proof that it occurs and removing the file fixes it:

[https://github.com/libretro/px68k-libretro/issues/98](https://github.com/libretro/px68k-libretro/issues/98)

### What device are you using?
RG353M

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen
2.0 (05112023)

### Anything else you'd like to include that may help to review this issue or feature request?

zomgugoff, 2023-10-17:
There is not a core option for ejecetion. The post is referring to the emulator menu accessed via L2. But when when trying to load a .m3u to start a game, it's a pain to do this ejection process. After ejecting, you have to use the terrible file browser in this menu to go reinsert the image file, which defeats the purpose of the m3u.

I added a rm command to my command line after reporting this and I haven't had a problem since.

christianhaitian, 2024-01-30:
Closing as this is something I will not accommodate at this time.  Hopefully libretro will resolve the issue at the core level if not done so already.

closed 2024-01-30 by christianhaitian