christianhaitian/arkos #717: Performance issue (RG535V) (05112023 release)

MetroidFusion, 2023-05-15:
Hi

When booting up the device and starting an original Game Boy game (e.g. Super Mario Land), I'm experiencing low performance/lag. The only way to solve this issue is by booting up a higher end system (e.g. Game Boy advance/mGBA), exiting the game, and then booting up the original Game Boy game again. When rebooting the device, I have to go through this procedure once again. It seems like the device is stuck in some kind of 'low performance/low power' mode, when playing original Game Boy games (and maybe other low-end systems) immediately after the initial boot. I have not experienced this on previous builds.  

/M

christianhaitian, 2023-05-15:
Maybe related to the issue reported here https://github.com/christianhaitian/arkos/issues/713.  I'm still investigating how to replicate this issue.

MetroidFusion, 2023-05-15:
I have tried exiting several gameboy games and reloading them. Unfortunately, the performance issue persists. It seems like the system need some kind of 'handshake' from more demanding software (e.g. gba emulation) for the CPU to perform as you would expect.

lauren7ino, 2023-05-17:
RG353M
Yesterday I installed and tested many PS1 games and found the graphics performance of Duckstation/Swanstation perfect at 2x resolution on Colin McRae Rally.
Today, I found that can't use any of these 2 emulators on the same games, even at 1x resolution. They can't reach 60 fps on Colin McRae Rally, and yesterday there was performance to spare.
I checked all settings twice, software/OpenGL rendering, everything. Even reset the Core settings to no avail.

christianhaitian, 2023-05-17:
> RG353M Yesterday I installed and tested many PS1 games and found the graphics performance of Duckstation/Swanstation perfect at 2x resolution on Colin McRae Rally. Today, I found that can't use any [...]

Between the day before and today, did you reboot your device or was it just put to sleep between sessions?

lauren7ino, 2023-05-17:
I shutdown and left it charging overnight.

christianhaitian, 2023-05-17:
Disable Power Saver Mode

christianhaitian, 2023-05-17:
Not having a TimeZone set could certainly cause that.  However, Enhanced Power Saver mode may cause that as well.  It's primarily designed to reduce functions of emulationstation but it could be reducing performance of the OS as well.

christianhaitian, 2023-05-18:
Please see if this performance issue has been resolved with today's update.  Be sure a timezone is set in the os and ensure that power saver mode is disabled as it may cause performance issues in the OS as well.

lauren7ino, 2023-05-18:
Time Zone restore seems to be fixed.
Performance issues, not so sure. I'm giving up DuckStation/SwanStation and trying other PSX emulators...

MetroidFusion, 2023-05-18:
Update: I just updated to the newest release (05172023). Unfortunately, the performance issue with gameboy games (/cores) persists. However, I have one interesting observation: When I manage to get a game (e.g. Super Mario Land) running with good performance (by booting up a higher end system, exiting the game, and then booting up the original gameboy game again), and THEN putting the device to deep sleep (which now works, nice!), the low performance 'returns'. Maybe there is a connection between the way the system 'wakes up' and its performance profiles, when launching low end systems.

christianhaitian, 2023-05-20:
This issue should be resolved with today's update.  Please update and confirm when you have a chance.  Thank you

MetroidFusion, 2023-05-21:
> This issue should be resolved with today's update. Please update and confirm when you have a chance. Thank you

I can confirm that the issue has been solved with the newest release (051923) - the system perform as expected now. Thank you very much! :)

christianhaitian, 2023-05-21:
Awesome.  Enjoy!

closed 2023-05-21 by christianhaitian