christianhaitian/arkos #722: Suggestion

lauren7ino, 2023-05-18:
Is it too far fetched to provide a CPU and GPU benchmark tool on the Tools section?
Maybe could help in performance issues, and serve a greater purpose as a way to compare different console models.
And what about a System Info dump?
Thanks

christianhaitian, 2023-05-18:
The benchmark tool is a fair request that I'll consider.  I won't bother with a System Info software/hardware dump as it's just static information about the device that most will already know.  Software info is already provided from the start menu if you scroll down to the bottom where it shows Distro Version.

lauren7ino, 2023-05-18:
Related question: Since these devices have almost no cooling, do you know if an increase in temperature causes the CPU to automatically lower the cpu/gpu clock? (to keep the temperature within safe limits)

christianhaitian, 2023-05-18:
That is possible but this chipset seems to run fairly cool until you push it for awhile with higher end emulation like psp or dreamcast.  "Awhile" meaning like 40 minutes or more.

lauren7ino, 2023-05-19:
I would say that this is the ideal time to launch the Benchmark tool.
You have been tweaking the power saver/sleep modes and maybe it’s something GPU/OpenGL related.
A CPU benchmark/burn tool and an OpenGL tool would be very useful to unfold this mistery.

lauren7ino, 2024-01-30:
I've tried to find a Demo that could be run,

lauren7ino, 2024-01-30:
Back to this old thread, that probably should be closed.

Anyway, I thought if one of these demos (https://demozoo.org/platforms/30/) could be used to make some kind of benchmark, by monitoring the FPS or time, to compare the performance of engines and shaders, but I'm at loss if that's even feasible.

I've tried this one:
https://demozoo.org/productions/67816/
... but my RG353M keeps it ~60 fps, so it's not heavy enough for a benchmark.

closed 2024-01-30 by christianhaitian