christianhaitian/arkos #725: Request - Touch Screen support

lauren7ino, 2023-05-20:
Hi

I was wondering it it's possible to add a touch screen driver to ArkOS that acts as a mouse?

I have 2 platforms ScummVM and Amiga which use mouse as input, a console that has touch screen (RG353M), but activating Touch Screen support in Reproach has no effect.

I found threads from 6 years ago that show that RetroArch supports in Android...
https://forums.libretro.com/t/mouse-support-added-to-android-input-driver-feedback-appreciated/8982

Thanks

christianhaitian, 2023-05-20:
Touchscreen support is currently not supported in Linux at this time.  Perhaps it will be enabled in the future for use in emulators.

closed 2023-05-20 by christianhaitian

lauren7ino, 2023-05-20:
Are you sure? I see no lack of references to Rockchip, Linux and Touch Screen

https://www.google.pt/search?q=%2Brockchip+cpu+touch+screen+support+linux

lauren7ino, 2023-05-20:
Found a few details ...

The touch screen panel used in the RG353P (RG353M too?) seems to be the Hytron CST 340.
https://www.google.pt/search?q=Hynitron+CST340+linux

Sorry if I'm being inconvenient. I will shut up now.

christianhaitian, 2023-05-20:
We've been testing it with the current kernel and it doesn't work in Retroarch or NDS.  Only Kodi recognizes the touch screen on the Linux side.

lauren7ino, 2023-05-20:
Thank you for your efforts 👍🏼

christianhaitian, 2023-05-20:
A community member by the name tomtomp has been experimenting with it.  If you have a RG353V, you can download the kernel and dtb from this file and try it yourself.  If your interested in more info about it and where he's at with his experimentation on it, you can try contacting him on the Retrogame Handheld Discord in the RG353V-VS channel.  The invite is https://discord.gg/retro-game-handhelds-529983248114122762

https://github.com/anaximan/rg353v-linux/raw/main/build/rg353v_linux_touch.zip

lauren7ino, 2023-05-21:
Thank you for the invite 😊
That's quite an active community! (I didn't knew it existed)
Replacing the files on my boot drive makes the console not to boot. Led lights up but nothing else AFAIK. Maybe I did something wrong.
I have a 353M, so maybe that's also the reason.
I think the touch screen support OR mouse/trackpad emulation using the touch screen is something that many of us dream about, but specially me on the Amiga/ScummVM games.
Thank you for your support.

christianhaitian, 2023-05-21:
It worked for me.  Make sure you used the files from the arkos folder and rename the included dtb file to rk3566-OC.dtb

lauren7ino, 2023-05-21:
Ah, thank you for the tip. I had to remove the -touch part of the file name and now it works!
Wow! I'm not going back to the regular kernel! :)

lauren7ino, 2023-05-21:
Works in Kodi and Amiga Standalone and  in the settings menu.
In Amiga itself (standalone) the mouse cursor behaves erratically on the vertical axis, so this leads me to believe that it's in fact a touch emulation and not a mouse/trackpad emulation.
Anyway, in my limited testing see no other inconvenience in this kernel.

I believe it should be freely distributed, to see if anyone developer knows how to fix the touch support in RetroArch and ScummVM.

lauren7ino, 2023-05-21:
Feels to me this is missed opportunity to jump ArkOS forward to the next level.

Maybe launch a new branch of ArkOS (ArkOS-Touch?) dedicated to touch enabled devices, that distributed the new Kernel files, touch enabled ScummVM (touch enabled RetroArch as soon is fixed) and start a movement to bring touch support to everyone?

christianhaitian, 2023-05-21:
I would if it had touch support for at least NDS but it doesn't yet.

lauren7ino, 2023-05-21:
(sorry my English, not my main language)

Maybe that's of those cases like:

You don't provide touch support because NDS doesn't have it,
and the nobody provides touch support for NDS because there are zero Linuxes with touch support.
You have the OS, you should give the first step.

Even if it's an Alpha version of Ark OS with zero maintenance, so the community gets hipped about touch support and start fixing emulators.
You can already provide support for Kodi, ScummVM (tomtomp Git) and a few more that mostly work (Amiga, etc.)

christianhaitian, 2023-05-22:
Interesting.  I will be testing this.

lauren7ino, 2023-05-22:
He also says that SummmVM support now is automatic since it runs over RetroArch touch. So, no need for a special compilation of Scumm.

lauren7ino, 2023-05-25:
Wow, that was fast! :D
I DM tomtomp and asked him to get me a touch version of RetoArch, which he agreed, be he instead released all his work on Discord.
---
About the "v2.0 (05/24/2023) Update (RG353M and RG353V units Only!)"
I deduce that all the RG353M/RG353V/RG353VS/RG353P use the same hardware (they share the same Anbernic firmware) so you may missing 2 models in that description.
---
I will test the new firmware as soon as I can. Did you compile a RetroArch32 version with touch support?
Thanks

christianhaitian, 2023-05-25:
The RG353VS has no touchscreen and I don’t support the rg353P as I don’t like the aesthetics of it.

christianhaitian, 2023-05-25:
I didn’t include an experimental build for retroarch32 since very few emulators need retroarch32 and none with touchscreen need that I can recall.

lauren7ino, 2023-05-25:
OK, understood.
I’ve is stalled ArkOS from scratch, updated and restored.
Everything I tested until now is fine.
Tomtomp says he’s going to work on RetroArch32 support.
UAE4ALL is the best Amiga emulator, and requires 32.
PUAE has too many problems. It can’t even restore Save States properly.

christianhaitian, 2023-05-25:
Oh i forgot about uae4all.

lauren7ino, 2023-05-25:
Just to avoid a duplication of efforts, I want you to know that Tomtomp sent me a RetroArch32 compiled with touch.
We are now investigating the direct touch issue with UAE4ALL.

lauren7ino, 2023-05-26:
Just to let you know Tomtomp's 32 bit RetroArch has passed the basic tests.
https://discord.com/channels/529983248114122762/1010962576047341700/1111751131572469802

anaximan, 2023-05-31:
Hi, I am the author behind the touch-enabled version of RetroArch. 
As the [PR](https://github.com/libretro/RetroArch/pull/15282) for the main RA repo is closing to a merge, I also created 
pre-built binaries for the latest code. It includes settings which allow 
toggling of features (settings -> input -> vMouse) and a new mode 
similar to a trackball. The binaries - aarch64 and armhf - are available 
for testing here: 
https://github.com/anaximan/rg353v-linux/raw/main/build/retroarch_2.zip
https://github.com/anaximan/rg353v-linux/raw/main/build/retroarch32_2.zip

christianhaitian, 2023-05-31:
Very cool.  Thank you

lauren7ino, 2023-05-31:
In these new versions I can no longer swap to direct mouse by triple tapping the top right corner?
I tried about 50 times with zero results.

anaximan, 2023-05-31:
Cool, I think that it should work out of the box on JelOS as well. The initial experiments 
I did with the touch panel were on JelOS, and I think I even have a distribution built. 
However, it is now quite obsolete. The modifications I made are compatible with their 
build system and are available here: 
https://github.com/anaximan/distribution
https://github.com/anaximan/rk356x-kernel
I am open to further cooperation to help getting it work on JelOS.

@lauren7ino sorry about that, it got disabled when I added automatic option updates. 
Here are updated executables with the fix: 
https://github.com/anaximan/rg353v-linux/raw/main/build/retroarch_4.zip
https://github.com/anaximan/rg353v-linux/raw/main/build/retroarch32_4.zip

lauren7ino, 2023-05-31:
Thanks, I’m commuting and I can’t test the executables.
Which combination is settings do I need to start in direct mouse touch mode every time?

christianhaitian, 2023-05-31:
> Cool, I think that it should work out of the box on JelOS as well. The initial experiments I did with the touch panel were on JelOS, and I think I even have a distribution built. However, it is now quite obsolete. The modifications I made are compatible with their build system and are available here: https://github.com/anaximan/distribution https://github.com/anaximan/rk356x-kernel I am open to further cooperation to help getting it work on JelOS.
> 
> @lauren7ino sorry about that, it got disabled when I added automatic option updates. Here are updated executables with the fix: https://github.com/anaximan/rg353v-linux/raw/main/build/retroarch_4.zip https://github.com/anaximan/rg353v-linux/raw/main/build/retroarch32_4.zip Otherwise, you can swap in the RetroArch settings and save it on per-core basis.

Ok.  You can also join the JelOS Discord here to further review with them: https://discord.gg/seTxckZjJy

christianhaitian, 2023-06-02:
Touch input seems to work better with the mednafen emulator for games like Hogan's Alley and Duck Hunt for NES.  Works like a touchscreen instead of a mouse.