christianhaitian/arkos #729: Error in certain ScummVM games with the new experimental touch support

jqno, 2023-05-26:
### Describe the Issue (If applicable)

With the new experimental touch support, Grim Fandango on ScummVM starts verifying game files, and ends up giving an error message: `ERROR: Could not open shader grim_background.vertex!!` If I disable touch support again, the game starts up fine.
I tried another ScummVM game (Day of the Tentacle), and there everything just works like a charm.

### How can the issue be reproduced? (If applicable)

* Enable experimental touch support
* Start Grim Fandango
* Wait a bit for the game files to be verified

### What device are you using?

RG353M

### What version of ArkOS are you on?

2.0 (05242023)

### Anything else you'd like to include that may help to review this issue or feature request?

I thought you might be interested in this kind of feedback given that the feature is still experimental. If not, just close the issue, it's no dealbreaker for me. Just happy that touch support is being worked on!

lauren7ino, 2023-05-26:
Q1: Does it work without the touch-version disabled?
Q2: If no, I don't have that game at the moment, but check you have everything up to date. ArkOS Updates + ScummVM updates:
In game: Start button -- Return to Launcher button -- Global options button --- Update Shaders button.

jqno, 2023-05-26:
As I mentioned, the game works fine with touch disabled.

I have updated the shaders from another ScummVM game, as you described (since I can't access the launcher from the error screen I described above), but the problem persists.

lauren7ino, 2023-05-26:
One last shot: Googling for "ArkOS" "Grim Fandango" I see references to "set the Game 3D renderer to software."
Good luck :)

jqno, 2023-05-26:
That does help in starting up the game, yes.

I did a little further digging, and I noticed that there's two directories containing ScummVM: `/opt/scummvm` and `/opt/scummvm.org` (when touch is enabled), or `/opt/scummvm` and `/opt/scummvm.touch` (when touch is disabled), and they get swapped around when enabling/disabling touch.

I noticed that the scummvm that belongs to touch-enabled, contains way less files than the 'original' scummvm. Among other things, the entire `extra/shaders` directory is missing. I copied it over, and suddenly the game starts up again (similar to when I "set the Game 3D renderer to software".)

jqno, 2023-05-26:
Thanks!

I noticed there's other files missing too. I don't know what they do, and some things are obviously different because of the added touch support, but here's a full list:

```
ark@rg353m:/opt$ diff -qr scummvm scummvm.touch/
Only in scummvm: COPYING
Only in scummvm.touch/: COPYRIGHT
Only in scummvm/extra: create-playground3d-data.sh
Only in scummvm/extra: create-testbed-data.sh
Files scummvm/extra/fonts.dat and scummvm.touch/extra/fonts.dat differ
Files scummvm/extra/freescape.dat and scummvm.touch/extra/freescape.dat differ
Only in scummvm/extra: gamecontrollerdb.txt.bak
Files scummvm/extra/kyra.dat and scummvm.touch/extra/kyra.dat differ
Only in scummvm/extra: monkey4-patch.m4b
Only in scummvm.touch/extra: nancy.dat
Only in scummvm/extra: Roland_SC-55.sf2
Only in scummvm/extra: shaders
Only in scummvm/extra: vkeybd_default
Only in scummvm/extra: vkeybd_small
Only in scummvm/extra: xeen.ccs
Only in scummvm.touch/LICENSES: COPYING.MKV
Files scummvm/NEWS.md and scummvm.touch/NEWS.md differ
Files scummvm/scummvm and scummvm.touch/scummvm differ
Only in scummvm/themes: default.inc
Only in scummvm/themes: fonts
Only in scummvm/themes: gui-icons.dat
Only in scummvm/themes: residualvm
Files scummvm/themes/residualvm.zip and scummvm.touch/themes/residualvm.zip differ
Only in scummvm/themes: scummclassic
Files scummvm/themes/scummclassic.zip and scummvm.touch/themes/scummclassic.zip differ
Only in scummvm/themes: scummmodern
Files scummvm/themes/scummmodern.zip and scummvm.touch/themes/scummmodern.zip differ
Only in scummvm/themes: scummremastered
Files scummvm/themes/scummremastered.zip and scummvm.touch/themes/scummremastered.zip differ
Only in scummvm/themes: scummtheme.py
Only in scummvm.touch/themes: shaders.dat
Files scummvm/themes/translations.dat and scummvm.touch/themes/translations.dat differ
```

christianhaitian, 2023-05-26:
Yea I went by the latest recommendation from scummvm's binary distribution recommendation here (https://wiki.scummvm.org/index.php?title=HOWTO-Release#Binaries).  Apparently it's incomplete.

christianhaitian, 2023-06-02:
This has been resolved with today's update.

closed 2023-06-02 by christianhaitian