christianhaitian/arkos #730: Freezing issue when trying to add new wifi connection on RGM351MP

molemitts, 2023-05-30:
### Before submitting your issue or feature request.  DO NOT REQUEST ASSISTANTS FOR PRELOADED IMAGES.  I do not support or condone the use of preloaded ArkOS images.  Preloaded meaning images that come with copyrighted roms preincluded.  Look for support from the creator or supplier of those preloaded images, not from here!  If a ticket is submitted for assistance for such images or if there's an indiciation that a preloaded image is what the support request is for, the issue ticket will be closed with no further response.

1. ***This is not the Stock OS that comes with your device, AmberElec, JelOS, or UnOfficialOS.***  ArkOS is a different distribution from those systems and will have differing features.  The only somewhat similarity is that they both use EmulationStation as a frontend.  Even with that, the versions of EmulationStation that's used in between ArkOS and these other distributions is different with different capabilities.  If there are features you prefer with the other available distributions, just use those distributions instead.  The choice is yours.

2. Have you reviewed the Frequently Asked Questions section available for your supported device in the wiki?

3. If your issue is related to being able to run certain emulators or ports, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

4. Have you reviewed the ***known issues*** section available at https://github.com/christianhaitian/arkos/wiki/Known-issues ?

### Describe the Issue (If applicable)
A clear and concise description of what the bug is.

When trying to add a wifi connection the console freezes at the moment the "+" is selected, the device freezes, requiring a hard shut down or reset. Using the TP-Link AC600 wifi adapter, JSAUX USB C male to USB 3.0 female, 100% battery, and without the charger plugged into the device while I'm trying to add the wifi connection. For a ~2 seconds the adapter blinks a green light and then stops blinking with no light on.

### How can the issue be reproduced? (If applicable)

Boot device, EmulationStation options menu, wifi, cursor over to "+", hit "A".
### What device are you using?

RGM351MP

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen

ArkOS 2.0 (03112023)

### Anything else you'd like to include that may help to review this issue or feature request?

Before making this post, I restored my OS microSD (PNY 64GB Elite-X Class 10 U3 V30) and reimaged it and the issue persisted. The freeze occurs whether or not the wifi adapter is connected, and happened after putting the arkosbackup.tar.gz file into the backup folder, following the steps included in the known issues workaround. I've since removed the workaround file from the backups folder.

molemitts, 2023-05-30:
Sorry--something I said in the original post was incorrect. It _does _not__ freeze when the "+" is selected without the adapter connected.

christianhaitian, 2023-05-30:
From what unit did you create the backup from?  Have you tried configuring your wifi before trying to restore a backup?

molemitts, 2023-05-31:
The backup I was restoring was the one you had linked as some sort of wifi fix on the "known issues" page. The link I used is [this one](https://github.com/christianhaitian/arkos/issues/124#issuecomment-751475195). I only did it out of desperation, really, as the issue in that thread was very dissimilar to the one I'm having, and the poster was using a different adapter and it was a driver fix, but you said it would "maybe improve wifi performance as well". I have tried to configure my wifi before restoring the backup, it was just something I was trying to fix it and thought I would mention I had read over the known issues page and tried that vaguely related fix.

molemitts, 2023-06-01:
I flashed AmberElec onto a spare microSD I had sitting around and _was_ able to make it work, so it's not a hardware issue I don't think. I'll probably reflash my main SD to AmberElec since the main goal was to get a wifi connection so I can set up PortMaster (thank you for your work on that, btw). Would it be helpful for you to continue to troubleshoot with me on ArkOS or is that sort of thing not worth investing your time? If needed, I can simply flash this spare SD with ArkOS once I get stuff transferred over to AmberElec on my main SDs.

christianhaitian, 2023-06-01:
Ahh that's a different adapter that's not officially supported in ArkOS.  The known working adapters are listed here: <https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---rg351mp#q-how-do-i-configure-my-wifi-connection-in-arkos>

molemitts, 2023-06-02:
I’m definitely facepalming right now. I looked at your FAQ to find an adapter to order but instead of clicking the link I typed it into my phone and thought I got the right one… Thanks for your help.

closed 2023-06-02 by molemitts