christianhaitian/arkos #731: No audio

lauren7ino, 2023-05-30:
Hi,
I installed ARkOS from scratch and updated everything, followed by a restore.
In a DOSBox-Pure game I had a freeze when I tried to force exit the game. I don't know exactly what I did, maybe I pressed Select+Start twice, not sure...
The system froze and I was forced to do a forced reboot.
After the reboot, ArkOS no longer outputs any audio.
Not even the Emulation Station navigation clicks.
Audio is at 100%.
Do you have any idea where I start to fix this problem?
I'm guessing the only fix is to install everything from scratch...
PS: Do you want an IMG of my SD Card for you to diagnose ?

christianhaitian, 2023-05-30:
Try pressing F+L3 and see if that fixes the issue.  If not, with the unit powered on, try plugging in a headphone into the 3.5mm audio jack then unplug it.

lauren7ino, 2023-05-30:
The headphone jack fixed it!!
You can't believe how many hours I've spent on this issue!!!
Thank you 3000! :-)
(Iron Man reference)

christianhaitian, 2023-05-30:
Did F+L3 not work?

closed 2023-05-30 by christianhaitian

lauren7ino, 2023-05-30:
Sorry, I pressed the DPad Left by mistake. Now I realized that L3 is actually the left joystick. I don't know if it works.
Do you think if would be nice if a Reset Audio in the advanced menu?

christianhaitian, 2023-05-30:
F+L3 is a Reset of audio