christianhaitian/arkos #732: Opentyrian doesn't work

VGkav, 2023-05-30:
Hello, I have installed a couple of ports and they run fine, for example Celeste Classic.

But now I installed OpenTyrian, the installer reported success but there is no entry for the game under Ports. I did restart EmulationStation and then the unit itself, I still don't see it. Any ideas?

Also, where are these ports located? Which directory?

VGkav, 2023-06-01:
Where is the ports folder exactly?

christianhaitian, 2023-06-01:
in the easyroms(roms) partition.  Same as other systems like nes and gba.

christianhaitian, 2023-06-05:
Any update on this?  Also, are you installing the ports via the portmaster tool located in the options/tools section?

christianhaitian, 2023-06-06:
Reading the instructions for the portmaster tool shown here https://github.com/christianhaitian/PortMaster#install-info

closed 2023-06-06 by christianhaitian