christianhaitian/arkos #734: RetroArch32 version on latest update

lauren7ino, 2023-06-02:
Hi
Today’s update is not distributing the latest RetroArch32 version, with the new Settings/Input Touch settings missing.
Thanks

lauren7ino, 2023-06-02:
Latest version here:
https://github.com/christianhaitian/arkos/issues/725#issuecomment-1555977944

christianhaitian, 2023-06-02:
It’s built from the merged commits from retroarch. So it’s newer than that.

lauren7ino, 2023-06-02:
OK.
Quick question: If the Experimental Touch Support is enabled, then we run the Update ArkOS, RetroArch and RetroArch32 gets replaced by the .touch versions, or we must toggle off/on Experimental Touch Support to guarantee that touch versions get used?

anaximan, 2023-06-02:
Hi, I checked out the latest update. It looks like the current RetroArch32-touch build is 
using an older version from the git. According to `--version`, it is "Git 976d7b8067". 
However, I am unsure as to why there even is any kind of touch support, since this 
commit is before my changes were merged into master. 

The 64b version ("retroarch.touch") is on "Git 243814da25", which is my latest commit 
in the touch PR. The touch function is fully implemented and includes the aforementioned 
input settings.

There might be some problem with the build using an older commit for RA32?

christianhaitian, 2023-06-02:
For retroarch, I ended up using you last retroarch touch enabled binary from here: <https://github.com/christianhaitian/arkos/issues/725#issuecomment-1570708793> because when I built it using the same latest commit from retroarch upstream, touch did not work.  Touch worked using that same Git 976d7b8067 for the retroarch32 build though.

christianhaitian, 2023-06-02:
> OK. Quick question: If the Experimental Touch Support is enabled, then we run the ArkOS Update, RetroArch and RetroArch32 gets replaced by the .touch versions, or we must toggle off/on Experimental Touch Support to guarantee that touch versions get used?

I accounted for that in the last update so if you're touch enabled, it updates to the latest provided touch enabled build for that.

anaximan, 2023-06-02:
> For retroarch, I ended up using you last retroarch touch enabled binary from here: [#725 (comment)](https://github.com/christianhaitian/arkos/issues/725#issuecomment-1570708793) because when I built it using the same latest commit from retroarch upstream, touch did not work. Touch worked using that same Git 976d7b8067 for the retroarch32 build though.

Hmm, I wonder why touch didn't work when you compiled the upstream. Did you define `UDEV_TOUCH_SUPPORT` 
during the build?

christianhaitian, 2023-06-02:
Doh!  No.  Why would it work with retroarch32 though?

christianhaitian, 2023-06-02:
Awesome.  Please do PR it and thank you again

anaximan, 2023-06-02:
Just to be sure, I tested retroarch32 as well. After adding the flag, everything seems 
to work correctly. I submitted the PR to the core repo. I noticed that you already fixed 
the problematic patch, so the only change is the `retroarch.sh` build script.

christianhaitian, 2023-06-02:
This has been resolved with today's update.  Thank you both again.

closed 2023-06-02 by christianhaitian

lauren7ino, 2023-06-22:
Hey,
Sorry to appear in this thread, but I can't find any other way to contact you.
I've been away for 2 weeks on vacation (yeah!) and I'm ready to resume testing ArkOS and Touch controls.
Do you have any forecast for the release of the next ArkOS update?
I'd like to start on a fresh version :)
Thanks

christianhaitian, 2023-06-22:
Fresh image or next update?  Next update will be somewhere within the next couple of weeks.  Fresh image I'm not planning for awhile.

lauren7ino, 2023-06-22:
Update. Thanks!