christianhaitian/arkos #735: Deep sleep (no longer?) works

lauren7ino, 2023-06-02:
RG353M

Here's a situation that I can repeat over and over:

Turn on the console from a shutdown.
Got to a game list on Emulation Station, select a game that auto-plays the game video after 2 seconds.
When the game video is playing, turn off the console (deep sleep).
Wait 5 seconds.
Turn on, then one of 3 things usually happen:
- Emulation Station crashes and reboots.
- The console is frozen.
- The console seems responsive, but when we use the DPad to go to the next game, the console freezes.
-----------------

- Doesn't seem to happen on Emulation Station main screen, so it may be related to playing game videos.
- Also doesn't happen with screensaver videos (I guess that's because the act of pressing the power button turns off the screen saver before sleep).
- I've cycled the Deep sleep off/on, just to make sure.
- Not sure for how long this problem exists.

Thanks

lauren7ino, 2023-06-02:
Edit: On Light Sleep, the same video playing situation above, Emulation Station reboots every time it wakes up.

christianhaitian, 2023-06-02:
disable power saving.  Emulationstation rebooting is a workaround for when videos are playing in emulationstation before it goes to sleep.  If emulationstation didn't reboot, it would freeze on wake.  If you want to reduce the possibility of emulationstation freezing when the unit is put to sleep, go back to the main menu then put the device to sleep or disable video screensaver.

lauren7ino, 2023-06-03:
Power saving off doesn't change anything. When wakes up, the console seems responsive (resumes playing the video), but when we use the Dead to go to the next game, the console freezes.

christianhaitian, 2023-06-03:
Then that’s the only one compatible with your device.  Not much we can do about that for now as some devices can deep sleep and some can’t.

closed 2023-06-03 by christianhaitian