christianhaitian/arkos #736: Please add .D71 and .D81 extensions to ES C64 extension lists

zomgugoff, 2023-06-05:
### Before submitting your issue or feature request.  DO NOT REQUEST ASSISTANTS FOR PRELOADED IMAGES.  I do not support or condone the use of preloaded ArkOS images.  Preloaded meaning images that come with copyrighted roms preincluded.  Look for support from the creator or supplier of those preloaded images, not from here!  If a ticket is submitted for assistance for such images or if there's an indiciation that a preloaded image is what the support request is for, the issue ticket will be closed with no further response.

1. ***This is not the Stock OS that comes with your device, AmberElec, JelOS, or UnOfficialOS.***  ArkOS is a different distribution from those systems and will have differing features.  The only somewhat similarity is that they both use EmulationStation as a frontend.  Even with that, the versions of EmulationStation that's used in between ArkOS and these other distributions is different with different capabilities.  If there are features you prefer with the other available distributions, just use those distributions instead.  The choice is yours.

2. Have you reviewed the Frequently Asked Questions section available for your supported device in the wiki?
Yes

3. If your issue is related to being able to run certain emulators or ports, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?
Yes

4. Have you reviewed the ***known issues*** section available at https://github.com/christianhaitian/arkos/wiki/Known-issues ?
Yes

### Describe the Issue (If applicable)
A clear and concise description of what the bug is.
The C64 core supports .D71 and .D81 file formats. Can you add them to the C64 extension list?

### How can the issue be reproduced? (If applicable)
Put .D71 or .D81 files in roms/c64, they are not detected in ES.

### What device are you using?
RG353M

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen
2.0 (06022023)

### Anything else you'd like to include that may help to review this issue or feature request?

christianhaitian, 2023-06-05:
Sure.  I'll add it in the next update.

christianhaitian, 2023-06-23:
This has been accommodated with today's update.  Just update from the options menu/section

closed 2023-06-23 by christianhaitian