christianhaitian/arkos #740: ArkOS not booting on the new Powkiddy RK2023 „pure black“ wifi model

waeggyfr, 2023-06-08:
device: Powkiddy RK2023 pure black wifi model
OS: ArkOS build 05192023 for RK 2023

issue:
The device is booting up until the screen „ArkOS 2.0 (xxxxxx) . . . .“ then the screen turns into black.
After that the device is not booting anymore even with the original SD card.

workaround: 
Disconnecting the battery of the device for few seconds will let the RK2023 boot up with the original SD Card.

observations:
The SD card delivered with the wifi model is compatible with no-wifi RK2023, but not vice versa
Error: Unable to find UUID-2003-2745, powering off and on should correct it
Tested that already and it is not working.

help:
To solve that issue i will provide the image of the SD card or sending the device within european countries

christianhaitian, 2023-06-08:
Were you using a fresh install of the image or an existing sd card?  If so, did you have 2 sd cards setup?  If so, you have to have both sd cards in to boot.  That issue seems like you didn't have the second sd card inserted at the time you were testing.

waeggyfr, 2023-06-09:
Tested:
1) fresh install with the build 05192023 shows just the CRT ArkOS bootlogo then black screen

2) used an existing SD card of the RK2023 device (single SD) shows ArkOS 2.0 (xxxxx) . . . . Then black screen

3) used an existing SD card (two cards setup) shows the same result like 2)

always stucks in a „boot loop“ and just disconnecting the battery recovers the device.
I tried it today with another wifi model, since I ordered three of them.

[image]

christianhaitian, 2023-06-09:
Ok.  If I receive this unit I will test and see if I can resolve this issue.  Thanks for reporting it.

christianhaitian, 2023-06-12:
Hi, I've received a report that booting upstream JelOS, then doing a proper shutdown will then allow ArkOS to boot on this device.  Can you try this and let me know your results?

Thanks,

waeggyfr, 2023-06-12:
Tested:

1) fresh installation of JelOS (build 20230610)
-> running without any issues, wifi is working out of the box

2) fresh installation of ArkOS RK2023 build 05192023 -> ArkOS installation & booting succeeded

issues:
- wifi is not working (driver problem?), tried manually creating the wifi connection (RetroArch, nmtui, direct edit) without success.

- FileManager window not centered, need to be set slightly upwards

[image]
[image]
[image]
[image]
[image]
[image]
[image]

christianhaitian, 2023-06-12:
Thank you for confirming.  My understanding is that the rg353m image works with the internal wifi.  I should be able to address the wifi once I have my unit.

nsrosenqvist, 2023-06-18:
I have a similar issue, the device gets stuck on the boot logo. Subscribing for updates regarding this issue.

christianhaitian, 2023-06-18:
Just waiting to receive my unit

Smythius, 2023-06-21:
Same issue with mine. Booted up once with original SD. attempted to flash ArkOS on SanDisk 512GB and couldn't get past the ArkOS splash screen. tried original SD. black screen. flashed Samsung 64GB SD with latest JelOS and Got screen to turn on at least. pressed reset and have figured upon powering on it returns a blank screen but will then load after pressing the restart button.

avicario96, 2023-06-25:
Hey guys, just wanted to add my findings when tinkering with Arkos firmware. As @christianhaitian mentioned that the wifi chip is similar to rg353m, What I found is if you write ArkOS_RG353M_v2.0_05042023.img to sdcard and then replace your boot partition files from ArkOS_RK2023_v2.0_05192023.img it does in fact work fine with wifi, I was able to connect and perform latest ota update for it without issue.

Another thing I noticed with my device is that it intermittently locks up, I believe its related to an incompatible dtb file as replacing it with the one from stock os fixed stability but caused some strange behavior with the bootup process, it sometimes requires a reset to fully start. Here's the [stock dtb file ](https://mega.nz/file/NHBHmBaZ#yDWU_gEPS0XpaeyR1FZxrW0XI4M1C3Ng2QBvIqnM81U)

Smythius, 2023-06-26:
How does one replace the boot partition files?

avicario96, 2023-06-26:
> How does one replace the boot partition files?

You would have to mount the boot volume by assigning a drive letter to it via disk management utility and then overwrite all the files from the Rk2023 image and also replace the dtb file. I'll save you the hassle, just download the premodified image [here](https://mega.nz/file/YCZ0hYRY#pvQ-aSzsS94n2g5gVd4jcY1Ln-WRtr164vbxC2BExUk).

jayjr1105, 2023-06-29:
FWIW I installed this to the new black model no issue and no tweaking whatsoever.  I used balena etcher and the ArkOS_RK2023_v2.0_05192023.img build.  I even used a USB ethernet dongle to update to the June22 build.  I can't toggle on wifi but otherwise it's working fine.

cndjonno, 2023-06-30:
I just installed 05192023 on wifi black without issue. Wifi does not work as noted previously. I look forward to you fixes @christianhaitian!

christianhaitian, 2023-07-03:
I received a report that the attached backup file I created should enable the internal wifi for this unit using the latest available image for the rk2023.

1.  Take the attached compressed file (arkosbackup.tar.gz) and put it in your roms/backup folder. Do not uncompress the contents! Just place the file as is in your roms/backup folder. If you don't have a backup folder in the roms folder, just create it. If you're using a 2 SD card setup on your device, this must be done from SD1(TF1 Slot).

2. Then go into the Emulationstation Menu, scroll to Options, hit A, then go to Advanced, hit A, then scroll all the way down to Restore Settings, hit A.

3. Once completed, reboot your device.

4. While loading, the device will reboot one more time on its own.

5. After the self reboot, the device will be ready for use.  Please post your results.

Thanks,

[arkosbackup.tar.gz](https://github.com/christianhaitian/arkos/files/11932977/arkosbackup.tar.gz)

christianhaitian, 2023-07-03:
New image with support for the internal wifi and bluetooth module is now available for download form the main wiki page.  Closing this issue as I received my pure black wifi model and able to confirm functionality with the included wifi and bluetooth modules are working with the newest image uploaded today.

closed 2023-07-03 by christianhaitian

nsrosenqvist, 2023-07-03:
> New image with support for the internal wifi and bluetooth module is now available for download form the main wiki page. Closing this issue as I received my pure black wifi model and able to confirm [...]

Thank you so much! I'll be testing this out tomorrow!

nsrosenqvist, 2023-07-10:
@christianhaitian I had it working for a while and suddenly I was back to the old behavior where it froze at the boot logo (it turns black after a little while). I tried reflashing a new SanDisk SD Card but I have the same issue still. How can I go about debugging this?

christianhaitian, 2023-07-10:
Are you using 2 sd cards?  If so, is it getting stuck at the arkos 2.0 with the dots moving?

nsrosenqvist, 2023-07-11:
@christianhaitian the graphical "arkos" splash is shown, and after a little while the screen goes black and I never see the ascii boot animation. I've tried this both without a second SD card inserted and also with one inserted. The image I'm using is the latest RK2023 one from 06222023.

christianhaitian, 2023-07-11:
Have you tried fully charging your device to see if that makes a difference?

christianhaitian, 2023-07-11:
@nsrosenqvist See if replacing the existing rk3566-OC.dtb file in the boot partition with the one in the attached zip file resolves the issue for you.

[rk3566-OC.zip](https://github.com/christianhaitian/arkos/files/12022149/rk3566-OC.zip)

nsrosenqvist, 2023-07-14:
@christianhaitian home again, just tried the file rk3566-OC file you sent but no luck. Will leave the device charging for a good while and see if anything changes. Is there any way to enable debug output if this doesn't work?

christianhaitian, 2023-07-14:
Usually only after you pass the ArkOS boot logo and get to the loading screen.  There you can press the volume down button and get verbose logging of activities.

nsrosenqvist, 2023-07-16:
@christianhaitian fully charging had no effect, and I can't seem to get JELOS working either, so I assume it must be some type of hardware issue. With JELOS I get a message saying "Unable to find UUID-...."

christianhaitian, 2023-07-16:
Did you try upstream JelOS?

nsrosenqvist, 2023-07-30:
@christianhaitian I tried it with https://github.com/JustEnoughLinuxOS/distribution/releases/tag/20230716, but no luck. Currently trying to get in touch with PowKiddy support.