christianhaitian/arkos #744: Feature request

nortxort, 2023-06-17:
Have you reviewed the Frequently Asked Questions section available for your supported device in the wiki?
Yes

### What device are you using?
Anbernic RG353M

### What version of ArkOS are you on?
2.0 (05042023)

Maybe i have missed it, if so, sorry for this request.
I cant find a way to search for games,. It would also be nice with a all games count. These were already part of the stock OS that came with my device, would be nice if arkos also had them included.

When i go to the "Options" menu it says "32 Games available" Should it not say "32 Options available" instead? i am on the "BS-THEME-FREPLAY" theme

christianhaitian, 2023-06-17:
While viewing a specific collection, you can get to the filter option by pressing “select”. If you want to search all games, you must enable the all games collection and the same rules apply for that collection

closed 2023-06-17 by christianhaitian

nortxort, 2023-06-18:
@christianhaitian So how do i enable the "all game collection"? I think i have been through all options, and i cant find it anywhere.

christianhaitian, 2023-06-18:
Press start>game collection settings>automatic game collections

then when es restarts, press y

nortxort, 2023-06-19:
Thanks a lot!