christianhaitian/arkos #745: Bugged ScummVM control mapping in ArkOS. There are Grim Fandango control issues

JMAccorsi, 2023-06-18:
### Describe the Issue (If applicable)
Right now Grim Fandango it's unplayable because of the controls (directions not working properly (right and left doesn't work), it's not possible to interact with objects, it's not possible to use the game menu to save etc).
It's impossible to set the commands for the game in the ScummVM menu because there's a constant input of "Alt Right" so I cannot edit the commands there ( you try to remap a command and it immediatly remap in "Alt Right"). 

### How can the issue be reproduced? (If applicable)
Start grim fandango with the default emulator and try to play the game.

### What device are you using?
RG353M

### What version of ArkOS are you on?
ArkOS 2.0 (06022023)

christianhaitian, 2023-06-18:
This is due to the force quit hotkey used for scummvm.  A fix is coming for that in the next update.

christianhaitian, 2023-06-23:
This has been resolved with today's update.  Just update from the options menu/section

closed 2023-06-23 by christianhaitian