christianhaitian/arkos #746: Feature Request:  ext4 on RG533V's secondary SD card

durkada, 2023-06-20:
It's my own fault for not reading the full motivation and ideas behind this OS, but, I migrated my roms to an ext4 partition  on the secondary SD card and switched it in the menu.  Worked great!  Until I reboot and, of course, it fails.

Eventually I dug in, read exfat was the the supported filesystem and migrated my games to that format.  Works perfectly fine, now.

But, as a linux user, it would be nice if ext4 could be supported natively.  I manage my rom collections with a reliance on symlinks, which exfat does not support.  Regardless, being a linux based system, it would be great if the secondary SD card could be the de-facto file system.

christianhaitian, 2023-06-21:
ext4 is a supported format for the secondary sd card.  You should be able to format it as ext4, do the switch to sd2, then copy your roms.

christianhaitian, 2023-06-21:
Ok.  My initial thought is that it could be a permissions issue of the sd card.  I'll check it out as soon as I can.

christianhaitian, 2023-06-22:
I can't recreate this issue.   ext4 works fine on my end on my device with ArkOS on a secondary sd card.  I was able to reboot  multiple times without issue.  Also worked fine from a cold boot.  Did you make sure to update ArkOS online after the initial boot?

[image]

closed 2023-06-24 by christianhaitian