christianhaitian/arkos #749: Add onscripter Emulator please

lcdyk, 2023-06-26:


christianhaitian, 2023-06-26:
I found a retroarch core this (https://github.com/iyzsong/onscripter-libretro).  I'll get it built.  If you can test and confirm it functions, I can get this added to ArkOS.

christianhaitian, 2023-06-26:
The core is now available in the central repo.  Just go into retroarch and go to core downloader, download the onscripter core, and try your content from there.  Let me know the results.

Thanks,

christianhaitian, 2023-06-27:
Looks like native controls are not working on the core.  We're looking to see how best to address that.

christianhaitian, 2023-07-06:
Core has been updated by the developer to support native controls.  You're welcome to test and confirm functionality again.

lcdyk, 2023-07-15:
> Core has been updated by the developer to support native controls. You're welcome to test and confirm functionality again.

I have been tested and found a problem。default.ttf could not be loaded causing the dialog font of the Chinese game to display incorrectly。Hope it can be improved, thank you

christianhaitian, 2023-07-15:
Did you have the default.ttf file in the same folder with the game?

lcdyk, 2023-07-16:
> Did you have the default.ttf file in the same folder with the game?

> Did you have the default.ttf file in the same folder with the game?

Yes, I have the default.ttf file in the same folder, but the Chinese dialog does not load successfully

christianhaitian, 2023-07-16:
What game is this so I can review the issue.

lcdyk, 2023-07-16:
> What game is this so I can review the issue.

https://drive.google.com/file/d/1NhIoHhsF7_O_ccAf2P-jB4dL1ZWXv-Ki/view?usp=sharing

christianhaitian, 2023-07-23:
This has been accommodated with today's update

closed 2023-07-23 by christianhaitian