christianhaitian/arkos #752: Switch to SD2 for Roms removes most options

OOaker, 2023-07-04:
ArkOS 2.0 on a RG353PS. After newly loading ArkOS everything is fine and works perfectly. However, once I use the script „switch to sd2 for roms“ almost all entries in the options menu disappear. Including all bluetooth functionality, update entry and so on. Basically only the wifi options remain.

OOaker, 2023-07-04:
Okay, the problem is not the SD2 script but simply that the file /opt/system/gamelist.xml is not up to date and does not reflect all the scripts in the folder. So if you force emulationstation to only load gamelist entries then most scripts vanish.

christianhaitian, 2023-07-05:
Correct. Simply don’t set that option as it doesn’t save any time on boot up.

closed 2023-07-05 by christianhaitian