christianhaitian/arkos #754: How to build the pcsx_rearmed libraries for an RG353M?

schusterro, 2023-07-05:
### Before submitting your issue or feature request.  DO NOT REQUEST ASSISTANTS FOR PRELOADED IMAGES.  I do not support or condone the use of preloaded ArkOS images.  Preloaded meaning images that come with copyrighted roms preincluded.  Look for support from the creator or supplier of those preloaded images, not from here!  If a ticket is submitted for assistance for such images or if there's an indiciation that a preloaded image is what the support request is for, the issue ticket will be closed with no further response.

1. ***This is not the Stock OS that comes with your device, AmberElec, JelOS, or UnOfficialOS.***  ArkOS is a different distribution from those systems and will have differing features.  The only somewhat similarity is that they both use EmulationStation as a frontend.  Even with that, the versions of EmulationStation that's used in between ArkOS and these other distributions is different with different capabilities.  If there are features you prefer with the other available distributions, just use those distributions instead.  The choice is yours.

2. Have you reviewed the Frequently Asked Questions section available for your supported device in the wiki?

3. If your issue is related to being able to run certain emulators or ports, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

4. Have you reviewed the ***known issues*** section available at https://github.com/christianhaitian/arkos/wiki/Known-issues ?

### Describe the Issue (If applicable)
There was a fix in pcsx-rearmed on master yesterday (https://github.com/libretro/pcsx_rearmed), which fixes a game for me.

_Could you please tell me how to compile this for the Anbernic RG353M, so that I can copy the .so file(s) to the device so I can test/confirm in the fix works fine?_

I've downloaded the git repo for it, but don't know what flags to use for the compiler.

Or, if not, an estimation on when this could come into ArkOS?

Thank you!

### How can the issue be reproduced? (If applicable)

### What device are you using?
RG353M

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen
ArkOS 06222023 with PCSX-ReARMed (r23l 4373e29)

### Anything else you'd like to include that may help to review this issue or feature request?

christianhaitian, 2023-07-05:
You can use the build recipe from here: https://github.com/christianhaitian/rk3566_core_builds/blob/master/scripts/pcsx_rearmed.sh

I don't rebuild that core to often as there usually seems to be some time of regression but I'll get a new build out today.

closed 2023-07-05 by christianhaitian

christianhaitian, 2023-07-05:
both loaded up just fine for me.  Good luck.