christianhaitian/arkos #755: volume/brightness adjustment overlay BUG

xclover0, 2023-07-08:
hello,

when adjusting the volume/brightness of the device while INSIDE the game, the overlay is not showing.
but when I'm only the home screen or game selection, it shows..

my device is RG351MP and using the latest build 6/22

christianhaitian, 2023-07-08:
Not a bug.  A limitation. Those overlays show in emulationstation only.  They don’t work while outside of the interface. No idea if it will be possible to include that capability in the future.

closed 2023-07-08 by christianhaitian

xclover0, 2023-07-08:
I guess it will be possible. I use UnOfficialOS before and overlays works inside the game
they are using emulationstation also (I think)

christianhaitian, 2023-07-08:
The overlays do not work inside any games on uOS either.  Same for JelOS or the stock OS.