christianhaitian/arkos #757: Bluetooth audio stops after a few minutes

JasonMiesionczek, 2023-07-17:
RG353V, 06222023, galaxy bud Bluetooth ear buds. I am able to connect them and use them when I first boot up my device, but after a few minutes of playing the sound stops and I have to turn Bluetooth off/on and reconnect the buds for sound to work again.

please let me know any other info you may need.

christianhaitian, 2023-07-17:
Do you have another pair of bluetooth headphones you can try to see if the issue happens there?

christianhaitian, 2023-07-29:
Any update on this?

christian-taillon, 2023-07-31:
I can confirm that this issue occurs on my rg353v with both Google Pixel Buds and Microsoft Surface Headphones using the Bluetooth connection.

JasonMiesionczek, 2023-07-31:
I switched to a pair if cheaper Bluetooth ear buds and those seem to work more consistently

christianhaitian, 2023-08-12:
Ok.  Closing this issue as I can't replicate the issue with the bluetooth headsets I have.

closed 2023-08-12 by christianhaitian