christianhaitian/arkos #759: RG353M external controllers don't work in standalone emulators (all ok with RetroArch/Emulation station)

thorias, 2023-07-19:
I have a RG353M with the latest Arkos release (great job by the way!) and I like to play games while it's connected to my tv and using the otg port with an external controller (xbox),
everything works perfect with all systems working with RetroArch cores based and in Emulation station, however with standalone emulators (duckstation, ppsspp, yabasanshiro, etc...) the external controller doesn't work, only the "integrated" one of the RG353M.

Is this expected or did I missed something? It'll be absolutely perfect if an external controller is connected and automatically the standalone emulator get properly configured to use it.

Thanks again for your great job for Arkos! I'm looking forward for the next release!

christianhaitian, 2023-07-19:
That is expected.  standalone emulators will most likely not work automatically with the controller.  There might be certain exceptions like the PPSSPP emulator where it might work depending on the controller.

closed 2023-07-19 by christianhaitian

thorias, 2023-07-19:
Ok thank you for your confirmation and please keep up this great job for Arkos!

Regards,