christianhaitian/arkos #763: No Stereo Sound on ArkOS - Anbernic RG353V

spamsool, 2023-07-24:
Stereo Sound is not working properly when using ArkOS on external headphones.

### Describe the Issue (If applicable)

When using ArkOS on Anbernic RG353V I cannot get stereo sound working properly. Left and right channels are somehow blended together, so part of the left channel can be heard in right channel and vice versa. Channels should be completely separated from each other in separate earpieces. They are not.
No problem with stereo when I am running Android.

The issue does not exist when running built-in Android on RG353V - the same file have completely separated left and right channel and they do not interfere with each other, and are not heard in the other ear-piece.

### How can the issue be reproduced? (If applicable)
You can hear in in the any stereo enabled games like Sonic on Sega Megadrive but it can also be clearly noticed while playing any left and right channel testing mp3 like for example: http://www.aoakley.com/articles/stereo-test.mp3 using Kodi player.

### What device are you using?

I am using Anbernic RG 353V

### What version of ArkOS are you on?
Latest version of ArkOS v2.0 (07/23/2023)

### Anything else you'd like to include that may help to review this issue or feature request?
No issues with sound when running Android. L R channels are perfectly separated.

Thanks very much for your help.

christianhaitian, 2023-07-24:
Yes there is an issue with the current manufacturer provided kernel for this unit that caused the right channel to not play audio through the single speaker that a workaround was implemented for that issue.  That workaround impacted plugged in headphones.  Hopefully the attached solution addresses it.  One caveat is that if audio is currently playing when plugging in or unplugging headphones, the audio may need to be stopped or paused, then resumed in order for the channel separation
[arkosbackup.tar.gz](https://github.com/christianhaitian/arkos/files/12148939/arkosbackup.tar.gz)
 to be correct.

**This is only for the rg353v/vs units only running ArkOS.  This is not needed for any other supported devices and will most likely have a negative impact on any other units this is tried on.**

1. Take the attached compressed file (arkosbackup.tar.gz) and put it in your roms/backup folder. Do not uncompress the contents! Just place the file as is in your roms/backup folder. If you don't have a backup folder in the roms folder, just create it.  If you're using a 2 SD card setup on your device, this must be done from SD1(TF1 Slot).

2. Then go into the Emulationstation Menu, scroll to Options, hit A, then go to Advanced, hit A, then scroll all the way down to Restore Settings, hit A.

3. Once completed, reboot your rg353v and test it.

Let me know your results when you've had a chance to test this.

spamsool, 2023-07-24:
Thank you very much for your quick reply.

I restored the backup with the file provided. 
Indeed now both left and right channels are played just right on my headphones.
However the side effect is what you mentioned - right channel is now silent when played on built-in internal speaker.
So I assume I cannot have both? Or maybe in future release there is a possibility to have an option to choose in the Options menu to enable either working both channels on internal speaker or proper stereo separation in the headphones?

If I want to have working both channels on the speaker is there anything I could do to restore the setting other than reflashing sd-card?
Thanks very much once again :)

spamsool, 2023-07-24:
Thank you very much for your quick reply.

I restored the backup with the file provided.
Indeed now both left and right channels are played just right on my headphones.

However the side effect is what you mentioned - When I unplug the headphones and I play the sound again on the speaker - the right channel is now silent when played on built-in internal speaker. (I unplugged the headphones and played the stereo-test file using Kodi which was still opened).

Then I tried to quit Kodi and went back to ArkOS and then started Kodi again - the speaker workaround starts to work and I can hear both channels on internal speaker (but it also blendes channels in the headphones).
Have not tried in games yet, but for sure I need to quit Kodi for the workaround to activate back and forth.

Maybe there could be and toggle in Options to choose the stereo sound mode?
Is the HDMI audio out also affected by this issue?
Thanks very much once again :)

christianhaitian, 2023-07-24:
Because it's using a .asoundrc file to route sound, unfortunately even manual toggle would exhibit the same issue.  The fix would be to correct the issue in the kernel which maybe corrected in the mainline linux kernel which I haven't started the process to transition to just yet.  I don't know if this issue impacts HDMI out.  I haven't tested that.

closed 2023-07-26 by spamsool