christianhaitian/arkos #765: RG353V Pro Controller not connecting

Cliffofthewild, 2023-07-29:
### Before submitting your issue or feature request.  DO NOT REQUEST ASSISTANTS FOR PRELOADED IMAGES.  I do not support or condone the use of preloaded ArkOS images.  Preloaded meaning images that come with copyrighted roms preincluded.  Look for support from the creator or supplier of those preloaded images, not from here!  If a ticket is submitted for assistance for such images or if there's an indiciation that a preloaded image is what the support request is for, the issue ticket will be closed with no further response.

1. ***This is not the Stock OS that comes with your device, AmberElec, JelOS, or UnOfficialOS.***  ArkOS is a different distribution from those systems and will have differing features.  The only somewhat similarity is that they both use EmulationStation as a frontend.  Even with that, the versions of EmulationStation that's used in between ArkOS and these other distributions is different with different capabilities.  If there are features you prefer with the other available distributions, just use those distributions instead.  The choice is yours.

2. Have you reviewed the Frequently Asked Questions section available for your supported device in the wiki?

3. If your issue is related to being able to run certain emulators or ports, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

4. Have you reviewed the ***known issues*** section available at https://github.com/christianhaitian/arkos/wiki/Known-issues ?

### Describe the Issue (If applicable)
My pro controller showed up but then wouldn’t connect on the Bluetooth manager. I have tried multiple times to pair but it will not even show up in the Bluetooth manager. 

### How can the issue be reproduced? (If applicable)
My pro controller will not show back up in the Bluetooth manager. 

### What device are you using?
RG353V

### What version of ArkOS are you on?
2.0

### Anything else you'd like to include that may help to review this issue or feature request?

christianhaitian, 2023-07-29:
What pro controller?  Xbox Pro?  Nintendo Switch Pro?  8bitdo Pro?

Cliffofthewild, 2023-07-29:
Nintendo Switch Pro controller.

christianhaitian, 2023-07-29:
Ok.  I don’t have this controller so I can not help with that.  Perhaps someone else can chime in on this.

christianhaitian, 2023-09-27:
Closing this ticket as this is not something I will work to resolve at this time.

closed 2023-09-27 by christianhaitian

Certifr1ed, 2023-11-02:
This worked for me thanks and credit to wepiii on reddit

For whatever reason, the bluetooth cannot find a device that it has previously connected to and then disconnected from. No amount of pairing attempts will get that device to show again in the "connect new device" menu.  

To fix, you need to go to settings/options depending on your theme, then enable remote services. go back to the main menu, click start and find your ip address. Go on a computer and input that ip address to connect to ARCOS file manager. Go to settings, then user management. Add a new user with the name "root" and scope "/". Do not type the parenthesis on root or /. Make sure to click "admin" for this new user, and also set your password. Once done, log out and log back in with user root and your password. Go to Var/Lib/Bluetooth and delete all the files in that folder. They will all be mac addresses. Log out, Disable remote services on your device, go back to main menu, click start, quit, restart emulation station. Go back to options/settings, bluetooth, connect to new device, and now your device will appear in the list.

christianhaitian, 2023-11-02:
Interesting,  I can update the Bluetooth script to do the same during the unpair/remove process.  Thanks for sharing this.

gabuzom, 2024-10-23:
I encountered this as well:

- an 8bitdo micro controller got at first recognized well
- then ceased to recognize OR appear when trying to look for new devices
- tried this dozen times, using various settings of the controller, even upgraded its firmware...
... was even starting to question the hw when i found this post.

deleted content of /var/lib/bluetooth (just mounting the 3rd partition on the microSD card on my laptop was enough) and voila!

so there seems to be a persistent issue with this..

lauren7ino, 2024-10-23:
This thread might explain why I cannot find my Xbox Controller after being removed, no matter how I try, I can't see it listed again.
If was a couple months ago. I gave up blaming Microsoft.
Since them I reinstalled ArkOS, I probably can see it again in the list, but can't try right now.