christianhaitian/arkos #766: [RG353 and variants] external controlers main profile on either usb or bluetooth while detected and disable integrated controller while connected on HDMI for standalone emulators

thorias, 2023-07-29:
Hello Christian,

Is it possible to create an external controller(s) some sort of profile (or script) while the RG353 is connected via HDMI/detect HDMI out to a TV, in order to be able to use an external controller (either in USB/Bluetooth) with standalone emulators mainly? Basically the Player 1 integrated controller will be disabled and the external controller(s) will take over if detected/available?

It'll be great for a full plug&play solution using the RG353(x) devices when it is connected to a bigger screen TV HDMI's while using mainly the standalone emulators that are providing superior performance/compatibility VS some cores in RA.

I hope I made myself clear...

Thank you,

thorias, 2023-08-01:
Hello Christian,

Do you think my query above is feasible?

Thank you,

christianhaitian, 2023-08-01:
Which standalone emulators?  For PSP, it's just a matter of adding the sdl bindings to the gamecontrollerdb.txt file.  For N64, it's much more complicated.

thorias, 2023-08-07:
Hello Christian,

If you have the setup/opportunity, please connect your RG353 HDMI to a TV and an external controller (via usb/bluetooth) and go through the standalone emulators as configured in by default in ArkOS and you'll quickly find out what I meant.

Thank you,

christianhaitian, 2023-08-10:
This is not how this will work.  Each standalone emulator has it's own particulars and some may need more work than others to support an external controller especially since there are many different makes and models of external controllers.  The primary focus of ArkOS is handheld mode.  External controller support is limited at this time.  It may improve in the future.

thorias, 2023-08-10:
Hello Christian,

No problem, I understand, but thank you for your time to look into this.

This topic can be closed,

Thanks a lot for your great work on ArkOS and please carry on!

closed 2023-08-10 by christianhaitian