christianhaitian/arkos #767: Battery warning

AlexKCN, 2023-07-31:
### Before submitting your issue or feature request.  DO NOT REQUEST ASSISTANTS FOR PRELOADED IMAGES.  I do not support or condone the use of preloaded ArkOS images.  Preloaded meaning images that come with copyrighted roms preincluded.  Look for support from the creator or supplier of those preloaded images, not from here!  If a ticket is submitted for assistance for such images or if there's an indiciation that a preloaded image is what the support request is for, the issue ticket will be closed with no further response.

1. ***This is not the Stock OS that comes with your device, AmberElec, JelOS, or UnOfficialOS.***  ArkOS is a different distribution from those systems and will have differing features.  The only somewhat similarity is that they both use EmulationStation as a frontend.  Even with that, the versions of EmulationStation that's used in between ArkOS and these other distributions is different with different capabilities.  If there are features you prefer with the other available distributions, just use those distributions instead.  The choice is yours.

2. Have you reviewed the Frequently Asked Questions section available for your supported device in the wiki?

3. If your issue is related to being able to run certain emulators or ports, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

4. Have you reviewed the ***known issues*** section available at https://github.com/christianhaitian/arkos/wiki/Known-issues ?

### Describe the Issue (If applicable)
A clear and concise description of what the bug is.

### How can the issue be reproduced? (If applicable)

### What device are you using?

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen

### Anything else you'd like to include that may help to review this issue or feature request?

Hello!  I would like to know how to disable the low battery warning, could you help me?

AlexKCN, 2023-07-31:
And with the low battery, audio is not coming out in games, could you help me?

christianhaitian, 2023-07-31:
What device are you using?

AlexKCN, 2023-07-31:
Gameforce CHI

christianhaitian, 2023-07-31:
Are you wanting to disable the low battery audio warning?  If so, I haven't included a method to do so just yet.

AlexKCN, 2023-07-31:
Understood, would you be able to add it in the next release if possible?  Please

christianhaitian, 2023-07-31:
That is my plan

AlexKCN, 2023-07-31:
Thank you very much Christian ✌🏻😁

xclover0, 2023-07-31:
> Are you wanting to disable the low battery audio warning? If so, I haven't included a method to do so just yet.

hey, can you also add female voice and ability to choose what voice (either male/female) would like to use? thank you

christianhaitian, 2023-08-03:
The ability to disable the verbal warning along with choosing a different American voice (including female) will be included in the next update

christianhaitian, 2023-08-12:
This feature has been added with today's update.  You can find it in the Start Menu>Sound Settings section.

closed 2023-08-12 by christianhaitian

AlexKCN, 2023-08-12:
Very thanks Christian 🤗🤗🤗