christianhaitian/arkos #769: Holding B on enter wifi password screen takes a screenshot and opens Kodi

RumblePen, 2023-08-06:
RG353M latest release. On the wifi password screen if you hold B button, Kodi will quickly open and close and a screenshot will be taken and stored in the root folder on sd1.

christianhaitian, 2023-08-06:
There's no process in ArkOS that can take a screenshot in anything unless you are in a retroarch emulator.  Even then, you have to map a button to take the screenshot.  Is this an ArkOS install you got from this git or are you using a preloaded image from some other source that may have altered how the OS works??

RumblePen, 2023-08-06:
Sorry about the accidental close, screen jumped on me.

This is the current 353m image from your git. First time using ArkOs, hadn't changed a single setting yet, went right to wifi, didnt know how to backout of the wifi password screen (I accidentally clicked the wrong network) so tried holding B.

christianhaitian, 2023-08-06:
It sounds like you hit many different buttons constantly and somehow went back to the ES menu, hit start, started Kodi, the exitted it out.  I'm curious what screen shot you found on sd1. I can't recreate what you experienced no matter what buttons I hold and press while on the wifi password screen.

christianhaitian, 2023-08-06:
I think I see what you did now.  With the Wifi password screen shown, if you hit select+start, the screen blinks and you have to select the X buttons on the bottom left hand corner of the screen in order to exit the keyboard which then leaves you with a black screen.  I hit start, then the A button then Kodi started.  I hit select+start to quit Kodi, then the ES menu showed back up.  I'll see if there's a way to force quit the keyboard when in the wifi menu.

christianhaitian, 2023-08-07:
The fix for this will be included in the next ota update.  Thanks for reporting the issue.

christianhaitian, 2023-08-12:
The fix has been added with today's update.

closed 2023-08-12 by christianhaitian