christianhaitian/arkos #770: ArkOS Quick Mode issue

EchoFiend2000, 2023-08-07:
Using Latest Release with RK2023 Wifi. Retroarch. 

When using safe shutdown hotkey again after booting a game via Quick Mode, instead of shutting down the device, the emulation station loading screen appears and the device loads as if it has just been turned on normally. 

To reproduce: Turn on Quick Mode, enter game, safe shutdown, restart device autoloading game, safe shutdown again. 

Overall the new Quick Mode feature is absolutely brilliant, thank you for everything you do!

christianhaitian, 2023-08-07:
What system and game was this happening with?  Was it a pico-8 game?

EchoFiend2000, 2023-08-07:
I intially encountered it with Super Nintendo & Super Mario World. 
Issue also present on Tetris (GBC) & Castlevania III (NES)
I've tried some other systems and the feature is working perfectly for PS1 and Dreamcast on Retroarch 32.
Hope this helps!

christianhaitian, 2023-08-07:
Ha.  Yea, I see that ES does load but then shutsdown.  It looks like what's happening is that the shutdown process doesn't finish in time before ES decides to try to load.  I'll see how best I can mitigate that.

christianhaitian, 2023-08-07:
Just noting that on next boot, the last game saves as I was shutting down does boot to the last savestate even after ES showed up and the shutdown finally completed.

christianhaitian, 2023-08-07:
Yep, I'm reproducing the issue consistently.  This will be great for testing a fix for that.

christianhaitian, 2023-08-08:
Hi,

I think I have a solution for this issue.  It seems to have resolved it on my end and hoping you can help me test it.

1. Take the attached compressed file (arkosbackup.tar.gz) and put it in your roms/backup folder. Do not uncompress the contents! Just place the file as is in your roms/backup folder. If you don't have a backup folder in the roms folder, just create it.  If you're using a 2 SD card setup on your device, this must be done from SD1(TF1 Slot).

2. Then go into the Emulationstation Menu, scroll to Options, hit A, then go to Advanced, hit A, then scroll all the way down to Restore Settings, hit A.

3. Once completed, reboot and test quick mode.

Let me know your results when you've had a chance to test this.

[arkosbackup.tar.gz](https://github.com/christianhaitian/arkos/files/12286388/arkosbackup.tar.gz)

christianhaitian, 2023-08-08:
Hi,

I believe I've resolved the issue.  It's been successful in my testing so far and hoping you can help me test this as well.

1. Take the attached compressed file (arkosbackup.tar.gz) and put it in your roms/backup folder. Do not uncompress the contents! Just place the file as is in your roms/backup folder. If you don't have a backup folder in the roms folder, just create it.  If you're using a 2 SD card setup on your device, this must be done from SD1(TF1 Slot).

2. Then go into the Emulationstation Menu, scroll to Options, hit A, then go to Advanced, hit A, then scroll all the way down to Restore Settings, hit A.

3. Once completed, reboot your device and test the quick mode.

Let me know your results.

[arkosbackup.tar.gz](https://github.com/christianhaitian/arkos/files/12287077/arkosbackup.tar.gz)

EchoFiend2000, 2023-08-08:
Hi, 

I've tested the quick mode with the fix and while it is no longer booting into Emulationstation as it was before it isn't turning off the device. Instead it is indefinitely showing a blank screen (screen is still on) with lit power led. 

If I tap the power button to sleep the device while it is like this the screen turns off as usual (led still on), tapping again to wake out of sleep causes Emulationstation to load. 

Also I'm not sure whether this is relevant now but just confirming what you said before about the last played game still autobooting even after Emulationstation loads and device shut down manually through the software. 

Please let me know if there is any more info I can give or anything else I can help test

christianhaitian, 2023-08-08:
Hmm...I didn't get a non shutdown issue.  What system and game did you test where you had that issue?

EchoFiend2000, 2023-08-08:
I've mostly been testing with Super Mario World (SNES) but Tetris (GBC) is also having the same issue. Castlevania SOTN (PS1) is still working normally.

christianhaitian, 2023-08-08:
Can you downloading and try restoring this backup from here again and let me know if this is still happening.
https://github.com/christianhaitian/arkos/issues/770#issuecomment-1668864537

EchoFiend2000, 2023-08-09:
I have downloaded and restored the backup again and unfortunately the blank screen non shutdown issue is still present.

EchoFiend2000, 2023-08-09:
I actually have ssh'd into the device before to compile the DLC for the Sonic Mania port following the guide linked in the wiki. 

https://github.com/romadu/RSDKv5-Decompilation/blob/master/README.md#building-on-device-with-arkos

Could that be causing the issue?

christianhaitian, 2023-08-09:
Oh did you enable developer mode to do this?

EchoFiend2000, 2023-08-09:
I thought I had but going back through the compile instructions it seems like I didn't ever enable Dev mode, no wonder I couldn't build it haha. I've just tried to enable dev mode and I got a bunch of error messages (I'm such an idiot) so I went the nuclear option and just reflashed my SD card from scratch. I redownloaded that back up you sent and now the Quick Resume mode works perfectly! I think I must have messed something up during my failed attempt at compiling the Sonic Mania DLC. I just want to say thanks for all your help and everything you do Christian, these devices wouldn't be anywhere near as enjoyable without the hard work of community Devs like yourself.

christianhaitian, 2023-08-09:
Awesome.  No Dev mode is not needed for this and could potentially introduce some unwanted effects.  Good to know that this is working well for you so far.  Please keep testing and if you get different results, please update me here.  Thanks again for reporting the issue and helping to test this fix.

christianhaitian, 2023-08-12:
This fix has been added with today's update.

closed 2023-08-12 by christianhaitian

arnomolly, 2023-08-22:
Hi, I'm still having this issue.
I've only updated to the latest ArkOS yesterday from a much older build (couple of months earlier).
Never used dev mode or SSH'ed. Tried the Quick mode for the first time (so presumably after the fix) and immediately ran into the issue where closing a game gave me an endless blank screen. Almost went and reflashed my entire OS until I read @EchoFiend2000 was able to get into EmulationStation by tapping power twice.
Using a V1 screen RG353V. Hope this helps!

christianhaitian, 2023-08-22:
What games and systems did you test and encounter this issue with?

arnomolly, 2023-08-22:
Yesterday just with Pokemon Crystal on GBC. Just tested more cores and both versions of retroarch. All have this same issue. Let me know if you want me to test more.

christianhaitian, 2023-08-22:
What version of ArkOS are you on?  From Emulationstation, you should see it at the bottom of the start menu where it says distro version.  If it doesn’t say 08112023, do a OTA update. If it does, disable quick mode, then reenable quick mode again and see if that resolves the issue for you.

arnomolly, 2023-08-23:
I am running the correct version and have already enabled and disabled a couple of times over the past few days, no difference.

However, I have found the cause of the issue! It only occurs when my launchimage is set go ascii. Changed it to jpeg and it works like a charm. Changed it back and had the issue again, all on the same game. Any chance you can iron this out so I can keep using ascii?

christianhaitian, 2023-08-23:
Odd.  I'll check this out and see if I can replicate the issue.

christianhaitian, 2023-08-25:
Hi @arnomolly,

I was able to replicate the issue with quick mode and launchimage set to ascii.  I'm testing a fix that seems to work well for me so far.  It would be helpful if you could help me test this and confirm it fixes quickmode using the ascii launchimage for your as well.

1. Take the attached compressed file (arkosbackup.tar.gz) and put it in your roms/backup folder. Do not uncompress the contents! Just place the file as is in your roms/backup folder. If you don't have a backup folder in the roms folder, just create it.  If you're using a 2 SD card setup on your device, this must be done from SD1(TF1 Slot).

2. Then go into the Emulationstation Menu, scroll to Options, hit A, then go to Advanced, hit A, then scroll all the way down to Restore Settings, hit A.

3. Once completed, reboot your device and test the quick mode.

Let me know your results.

[arkosbackup.tar.gz](https://github.com/christianhaitian/arkos/files/12440452/arkosbackup.tar.gz)

arnomolly, 2023-08-27:
Hi Christian,

When I first tested it using your instructions, it worked. I booted into the game using Quick Mode, then succesfully quit to EmulationStation. Tested this twice using different cores. However, then I tried it with a Retroarch32 core and it did not work. After this, I haven't been able to get Quick Mode to work as intedend anymore when closing a game.

christianhaitian, 2023-08-27:
> Hi Christian,
>
> When I first tested it using your instructions, it worked. I booted into the game using Quick Mode, then succesfully quit to EmulationStation. Tested this twice using different cores. [...]

I just tested quick mode using retroarch32/pcsx-rearmed and Ape Escape as well as Lode Runner, both exited to ES without issue on my end using the ascii mode and jpeg mode.  Perhaps you were testing a different system on retroarch32 mode?

arnomolly, 2023-08-28:
Don't know what happened the first time, but I just re-applied the backup file. Everything including retroarch32 (I was using pcsx-rearmed as well) works fine now! Thanks a lot!

christianhaitian, 2023-08-29:
Awesome. Thanks for testing and confirming.