christianhaitian/arkos #772: USB DAC hotplug/volume control

zomgugoff, 2023-08-12:
### Before submitting your issue or feature request.  DO NOT REQUEST ASSISTANTS FOR PRELOADED IMAGES.  I do not support or condone the use of preloaded ArkOS images.  Preloaded meaning images that come with copyrighted roms preincluded.  Look for support from the creator or supplier of those preloaded images, not from here!  If a ticket is submitted for assistance for such images or if there's an indiciation that a preloaded image is what the support request is for, the issue ticket will be closed with no further response.

1. ***This is not the Stock OS that comes with your device, AmberElec, JelOS, or UnOfficialOS.***  ArkOS is a different distribution from those systems and will have differing features.  The only somewhat similarity is that they both use EmulationStation as a frontend.  Even with that, the versions of EmulationStation that's used in between ArkOS and these other distributions is different with different capabilities.  If there are features you prefer with the other available distributions, just use those distributions instead.  The choice is yours.

2. Have you reviewed the Frequently Asked Questions section available for your supported device in the wiki?
Yes

3. If your issue is related to being able to run certain emulators or ports, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?
N/A

4. Have you reviewed the ***known issues*** section available at https://github.com/christianhaitian/arkos/wiki/Known-issues ?
Yes

### Describe the Issue (If applicable)
A clear and concise description of what the bug is.

USB DACs work when the unit is started with one plugged in, but will not work if it's plugged in after booting. Also, when it is working, the system volume control does not function.

### How can the issue be reproduced? (If applicable)
-Plug in USB DAC while running, observe no sound coming from it
-Plug in USB DAC before booting (or reboot after plugging in), observe sound is coming from it
-Attempt to adjust volume with USB DAC producing audio, observe it does not work and the default output sounds like full volume

### What device are you using?
RG353M

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen
2.0 08112023

### Anything else you'd like to include that may help to review this issue or feature request?
The audio output from the RG353M is somewhat noisy. It sounds like bad shielding/isolation on IEMs. The DAC really blocks all of it. I am using a Rhodium USB-C dongle DAC.

christianhaitian, 2023-09-23:
This is an issue for all devices with this chipset.  Some worse than others.  I think I've got a solution for using a dac.  Hopefully you can help me test it.

1. Take the attached compressed file (arkosbackup.tar.gz) and put it in your roms/backup folder. Do not uncompress the contents! Just place the file as is in your roms/backup folder. If you don't have a backup folder in the roms folder, just create it.  If you're using a 2 SD card setup on your device, this must be done from SD1(TF1 Slot).

2. Then go into the Emulationstation Menu, scroll to Options, hit A, then go to Advanced, hit A, then scroll all the way down to Restore Settings, hit A.

3. Once completed, reboot your device and test your USB DAC.

Let me know your results.  Do note that when you plug your usb dac while your device is on, you will have to restart emulationstation so the sound can go through the usb dac instead of the external speaker.  The same is true when going from the usb dac back to the speaker as well.

[arkosbackup.tar.gz](https://github.com/christianhaitian/arkos/files/12706091/arkosbackup.tar.gz)

MetroidFusion, 2023-09-23:
Hi, just wanted to chime in. I have tested the solution above and can confirm that it works; my USB-C DAC is functioning after applying the file and volume can be  adjusted on the fly. However, even at maximum volume it not very loud. It might just be my DAC of course, even though it's actually pretty loud on other devices.

MetroidFusion, 2023-09-23:
Just tried disconnecting the USB-C DAC and sound returned to the speakers after restarting EmulationStation. However, I was not able to adjust the volume. I had to restart the device to fix this issue.

christianhaitian, 2023-09-23:
I noticed the same volume issue on my end with my dac and have a solution to address that.  The volume adjustment issue is probably related to the global hotkeys stopped working.  That happens on occasion.  There's a script called fixed global hotkeys in the options/advanced section that would fix that without requiring a reboot.  It needs an update as the confirmation key doesn't work on this chipset.  I have an update coming out shortly to address that and add this usb dac feature and improve the volume of the usb dac.

christianhaitian, 2023-09-23:
@MetroidFusion Thanks for your testing.

MetroidFusion, 2023-09-23:
That sounds great! I'll be sure to test out the DAC fix/implementation if you should decide to make it available as an 'arkosbackup.tar.gz'-file for testing purposes.

zomgugoff, 2023-09-24:
I'm glad someone was able to test today. In my own testing, I'm finding an issue on the published update. If the DAC is plugged in from boot and then unplugged in ES, starting a game will cause the system to freeze. If the DAC was unplugged from boot but is plugged in while ES is running, starting a game will result in audio out through the DAC, but will revert to the speaker in ES.

I know it was described to restart ES if plugging in after boot, but it might need to be documented that this freeze situation may occur.

christianhaitian, 2023-09-24:
I noted that the feature is in beta.  I'll probably include it in the wiki and point back to this issue for future documentation
Thanks @zomgugoff

MetroidFusion, 2023-09-24:
Just wanted to let you know that I have now tested published solution (09232023). Unfortunately, in my case at least, the audio is now extremely low when using a DAC. I can barely hear any audio coming from the device. This is compared to the 'arkosbackup.tar.gz'-file posted yesterday, which also resulted in somewhat low volume, but is was comparatively much higher than in the published release.

MetroidFusion, 2023-09-24:
Okay, so after disconnecting the DAC and rebooting the device, I actually also 'lost' the ability to proces audio through the speakers. The hotkeys for volume controls work, but no audio is coming from the speakers. I also tried to apply the 'old' fix from yesterday, but I still don't have any audio coming from the speakers . Maybe I will have to reflash :)

christianhaitian, 2023-09-24:
No need to reflash, just restore one of the backup packages below for the proper device:

RG353V/VS Units Only! ---->[arkosbackup.tar.gz](https://github.com/christianhaitian/arkos/files/12708537/arkosbackup.tar.gz)

RG353M/RG503/RGB30/RK2023 Units Only!--->[arkosbackup.tar.gz](https://github.com/christianhaitian/arkos/files/12708535/arkosbackup.tar.gz)

Make sure yo reboot and see if audio returns for your dac as well.  Should also be louder too.  Let me know your results.

MetroidFusion, 2023-09-24:
Awesome, that worked a treat! I actually reflashed the OS before your post, and even though I regained sound through the speakers by doing that, the audio from the external DAC was still extremely low. But with the above backup package the external DAC now performs as expected volume-wise, and the speakers work as well. Thanks! :)

christianhaitian, 2023-09-24:
Awesome.  Thanks for confirming.

christianhaitian, 2023-09-30:
The latest usb dac support has been included with today's update.

closed 2023-09-30 by christianhaitian