christianhaitian/arkos #775: What do the power saver modes do?

MoTem, 2023-08-22:
### Before submitting your issue or feature request.  DO NOT REQUEST ASSISTANTS FOR PRELOADED IMAGES.  I do not support or condone the use of preloaded ArkOS images.  Preloaded meaning images that come with copyrighted roms preincluded.  Look for support from the creator or supplier of those preloaded images, not from here!  If a ticket is submitted for assistance for such images or if there's an indiciation that a preloaded image is what the support request is for, the issue ticket will be closed with no further response.

1. ***This is not the Stock OS that comes with your device, AmberElec, JelOS, or UnOfficialOS.***  ArkOS is a different distribution from those systems and will have differing features.  The only somewhat similarity is that they both use EmulationStation as a frontend.  Even with that, the versions of EmulationStation that's used in between ArkOS and these other distributions is different with different capabilities.  If there are features you prefer with the other available distributions, just use those distributions instead.  The choice is yours.

2. Have you reviewed the Frequently Asked Questions section available for your supported device in the wiki?

3. If your issue is related to being able to run certain emulators or ports, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

4. Have you reviewed the ***known issues*** section available at https://github.com/christianhaitian/arkos/wiki/Known-issues ?

### Describe the Issue (If applicable)
Not sure where else to post this. I found no documentation on what the power saver modes do on arkos. I would love an addition to the FAQ section about it and the performance costs per setting. 

### How can the issue be reproduced? (If applicable)

### What device are you using?
RG353Vs

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen
v2.0 (08/11/2023)

### Anything else you'd like to include that may help to review this issue or feature request?

christianhaitian, 2023-08-22:
I haven't tested them much.  They came with with emulationstation and I initially thought they did nothing and were more geared towards PC users.  I've been hearing that they do make some differences.  What they are I'm not sure.  You're welcome to test them and report your findings here.  I'd be happy to include your findings in the FAQa and credit you with the findings.

MoTem, 2023-08-22:
Got it. I dug in further and found information on it here: https://github.com/RetroPie/EmulationStation/pull/172

My understanding is that it is beneficial on extremely low-powered battery devices, and of course totally useless on devices which are always plugged in. It does not affect game performance (I verified this for PSX emulation with no loss of fps on any setting) and is more for the emulation frontend, by reducing idle CPU usage. I tested playing around the menus with all modes enabled and it didn't effect performance of the front-end (observing the inbuilt fps counter) nor disabled any user-facing features like showing box art and playing preview videos with sound (tested on all the default themes). 

My suggestion is that keeping it enabled costs nothing to the user but is giving some unobservable (since I cant check CPU usage) benefit based on what other users are claiming. So keeping it at least at "default" should be safe.

MoTem, 2023-08-22:
Accidentally closed it.

christianhaitian, 2023-09-02:
Do any of the other modes impact sleep reliability?

MoTem, 2023-09-02:
Not that I have noticed. Just a disclaimer I use the deep sleep setting.

Lamcza, 2023-09-18:
> Do any of the other modes impact sleep reliability?

yes if you enable any of them you device will crash in deep sleep :D

christianhaitian, 2023-09-18:
Oh.  That may explain why I never have crashing issues in Deep Sleep.  I never change those.

MoTem, 2023-09-22:
I tested again and I agree that there is infact more instability in deep sleep when power saver modes are on. So I stand corrected, better to leave it off.

closed 2023-09-22 by MoTem

christianhaitian, 2023-09-22:
Good info and I'm compelled to remove that in a future update.