christianhaitian/arkos #776: Does deep sleep on rg353v work?

JulyIghor, 2023-08-24:
In OFW, when I press the power button, the device goes to sleep and the LED turns off.
However, in ArkOS, when I do the same, the LED stays on.
Is this the expected behavior? Is there a way to enable deep sleep in ArkOS similar to how it functions in OFW?
The battery drains much faster compared to OFW.

christianhaitian, 2023-08-24:
Depends on the device.  You can try to enable deep sleep in the options>advanced settings in ArkOS on your device and see if that works reliably for you.  The LED being on or off doesn't cause the drain.  It's the type of sleep.  my rg353 devices go into deep sleep fine but the results are not consistent for everyone's device.  good luck.

closed 2023-08-24 by christianhaitian

JulyIghor, 2023-08-24:
Thanks for the fast answer.

> Depends on the device. You can try to enable deep sleep in the options>advanced settings in ArkOS on your device and see if that works reliably for you
I tried all sleep modes of ArkOS, and they does not behave as it is done in OFW. 

I have a few question and it would be great if you can answer, so I can find a fix:

> It's the type of sleep
Which type of sleep mode is used in the official firmware of the RG353V when the LED turns off?

> my rg353 devices go into deep sleep fine
Does the LED turn off for you on that device in sleep mode?

JulyIghor, 2023-08-25:
Also I tried the command via ssh echo "mem" > /sys/power/state
And it goes to deep sleep for real, but it won't wake up via power button, any advice?

JulyIghor, 2023-08-25:
Also deep sleep mode isn't visible here, I think it is broken with rg353v in ArkOS

[image]

christianhaitian, 2023-08-25:
LED is not controlled by the type of sleep.  That is controlled in the kernel device tree (https://github.com/christianhaitian/RG353VKernel/commit/2cc8702d59755fd64d5cee2cef97118250d3d0e5).  If the official firmware truly does go into a deep sleep, then Anbernic may have addressed it in their official firmware and not shared with the community how they did this.  This is typical for them as of late.  Your device not waking from deep sleep is the common issue some folks have with their device.

JulyIghor, 2023-08-25:
> LED is not controlled by the type of sleep.  That is controlled in the kernel device tree (https://github.com/christianhaitian/RG353VKernel/commit/2cc8702d59755fd64d5cee2cef97118250d3d0e5).  If the [...]

Take a look at today's JELOS update, deep sleep started to work correctly

christianhaitian, 2023-08-25:
I checked with the developers.  There was no fix for deep sleep in JELOS for this chipset.

JulyIghor, 2023-08-25:
> I checked with the developers. There was no fix for deep sleep in JELOS for this chipset.

Ok, how can you explain this?

The LED indicates that deep sleep is active in JELOS, but in ArkOS the LED remains on constantly.

https://github.com/christianhaitian/arkos/assets/3390357/345878f9-2c31-4d39-a504-7cf428d59e4c

https://github.com/christianhaitian/arkos/assets/3390357/91f971fd-739b-4e38-b569-328c77ce8e33

christianhaitian, 2023-08-25:
Again, the led being active and whether a device goes into a deep sleep or light sleep has no correlation.  It's a deliberate choice I made to keep the led on while in sleep (whether deep or light) because I could not remember when I put my device to sleep or not without have an indicator.  The LED being on doesn't cause the excessive battery drain.

JulyIghor, 2023-08-25:
> Again, the led being active and whether a device goes into a deep sleep or light sleep has no correlation. It's a deliberate choice I made to keep the led on while in sleep (whether deep or light) [...]

Ok, thanks for your time and answers