christianhaitian/arkos #777: Ports not worked

quicksilvereu, 2023-08-25:
DEVICE : ODROID GO ADVANCE V2 
VERSION : 11.8.2023
I have ported quite a few games that worked.  Such as Ninja Turtles, Bleed 1 and Bleed 2 and other.  After the update it stopped working.  I think it is done by the SDL2 library, version 2.0.28.2, would there be a solution other than reinstalling the whole thing?  Thanks

christianhaitian, 2023-08-25:
Those ports are still in testing so no surprise a newer sdl could affect their functionality.  What you can do is manually edit the .sh file for each of those ports and force them to use the older sdl or change the sdl link to go back to the old version.

christianhaitian, 2023-08-25:
To change the SDL systemwide, you can run the following from ssh:

For 64bit:
`sudo ln -sfv /usr/lib/aarch64-linux-gnu/libSDL2-2.0.so.0.2600.5 /usr/lib/aarch64-linux-gnu/libSDL2.so`

For 32bit:
`sudo ln -sfv /usr/lib/arm-linux-gnueabihf/libSDL2-2.0.so.0.2600.5 /usr/lib/arm-linux-gnueabihf/libSDL2.so`

To do it on a per port basis, just edit the port execution line and add the following to the front of the execution:

`LD_PRELOAD= /usr/lib/aarch64-linux-gnu/libSDL2-2.0.so.0.2600.5`

For instance, for Ninja Turtles: you would change line 110 to be:
`LD_PRELOAD= /usr/lib/aarch64-linux-gnu/libSDL2-2.0.so.0.2600.5 $TASKSET mono ../MMLoader.exe MONOMODDED_${gameassembly} > >(tee ${gamedir}/log.txt) 2>&1`

quicksilvereu, 2023-08-25:
#!/bin/bash

if [ -d "/opt/system/Tools/PortMaster/" ]; then
  controlfolder="/opt/system/Tools/PortMaster"
elif [ -d "/opt/tools/PortMaster/" ]; then
  controlfolder="/opt/tools/PortMaster"
elif [ -d "/roms/tools/PortMaster" ]; then
  controlfolder="/roms/tools/PortMaster"
elif [ -d "/roms/ports" ]; then
  controlfolder="/roms/ports/PortMaster"
elif [ -d "/roms2/ports" ]; then
  controlfolder="/roms2/ports/PortMaster"
else
  controlfolder="/storage/roms/ports/PortMaster"
fi

source $controlfolder/control.txt
source $controlfolder/tasksetter

get_controls

gameassembly="Bleed.exe"
gamedir="/$directory/ports/bleed"
cd "$gamedir/gamedata"

# Grab text output...
$ESUDO chmod 666 /dev/tty0
printf "\033c" > /dev/tty0
echo "Loading... Please Wait." > /dev/tty0

# Setup mono
monodir="$HOME/mono"
monofile="$controlfolder/libs/mono-6.12.0.122-aarch64.squashfs"
$ESUDO mkdir -p "$monodir"
$ESUDO umount "$monofile" || true
$ESUDO mount "$monofile" "$monodir"

# Setup savedir
$ESUDO rm -rf ~/.local/share/MONOMODDED_Bleed
ln -sfv "$gamedir/savedata" ~/.local/share/MONOMODDED_Bleed

# Remove all the dependencies in favour of system libs - e.g. the included 
# newer version of FNA with patcher included
rm -f System*.dll mscorlib.dll FNA.dll Mono.*.dll

# Setup path and other environment variables
export MONO_PATH="$gamedir/dlls":"$gamedir/gamedata":"$gamedir/monomod"
export LD_LIBRARY_PATH="$gamedir/libs":"$monodir/lib":"$LD_LIBRARY_PATH"
export PATH="$monodir/bin":"$PATH"

export FNA3D_FORCE_DRIVER=OpenGL
export FNA3D_OPENGL_FORCE_ES3=1
export FNA3D_OPENGL_FORCE_VBO_DISCARD=1
regen_checksum=no

sha1sum -c "${gamedir}/gamedata/.ver_checksum"
if [ $? -ne 0 ]; then
	echo "Checksum fail or unpatched binary found, patching game..." > /dev/tty0 2>&1
	rm -f "${gamedir}/gamedata/.patch_done"
fi

# MONOMODDED files not found, let's perform patching
if [[ ! -f "${gamedir}/gamedata/.patch_done" ]]; then
	echo "Performing game patching..." > /dev/tty0 2>&1

	# Configure MonoMod settings
	export MONOMOD_MODS="$gamedir/patches"
	export MONOMOD_DEPDIRS="${MONO_PATH}":"${gamedir}/monomod"

	# Patch the ParisEngine/gameassembly files
	mono "${gamedir}/monomod/MonoMod.RuntimeDetour.HookGen.exe" "${gamedir}/gamedata/Bleed.exe" > >(tee ${gamedir}/monomod_log.txt) 2>&1
	mono "${gamedir}/monomod/MonoMod.exe" "${gamedir}/gamedata/Bleed.exe" > >(tee -a ${gamedir}/monomod_log.txt) 2>&1
	if [ $? -ne 0 ]; then
		echo "Failure performing first time setup, report this." > /dev/tty0 2>&1
		sleep 5
		exit -1
	fi

	# Mark step as done
	touch "${gamedir}/gamedata/.patch_done"
	regen_checksum=yes
fi

# Regenerate sha1sum checks
if [[ x${regen_checksum} -eq xyes ]]; then
	sha1sum "${gamedir}/gamedata/"Bleed.exe > "${gamedir}/gamedata/.ver_checksum"
	sha1sum "${gamedir}/patches/"*.dll >> "${gamedir}/gamedata/.ver_checksum"
fi

$GPTOKEYB "mono" &
$TASKSET mono MONOMODDED_${gameassembly} > >(tee ${gamedir}/log.txt) 2>&1
$ESUDO kill -9 $(pidof gptokeyb)
$ESUDO systemctl restart oga_events &
$ESUDO umount "$monodir"

# Disable console
printf "\033c" >> /dev/tty1

//////////
Could you please write down what I can edit?  Thanks

christianhaitian, 2023-08-25:
I did so already here: https://github.com/christianhaitian/arkos/issues/777#issuecomment-1693766791

quicksilvereu, 2023-08-25:
Wow, I used the first line of the sudo code and changed everything to 64bit via the ssh command and it works fine.  Thanks

closed 2023-08-25 by christianhaitian

Rivera177, 2023-10-01:
> To change the SDL systemwide, you can run the following from ssh:
>
> For 64bit: `sudo ln -sfv /usr/lib/aarch64-linux-gnu/libSDL2-2.0.so.0.2600.5 /usr/lib/aarch64-linux-gnu/libSDL2.so`
>
> For 32bit: [...]

It doesn’t work I tried it and it loaded the game up. But as soon as I tried to start the story mode i got booted to the menu of arkos postmaster. What should I do?