christianhaitian/arkos #778: RG353V sound does not output after boot until headphones are connected

fexofenadine, 2023-08-27:
### Describe the Issue (If applicable)
After booting the device into ArkOS, there is no sound output from the speakers until headphones are plugged in.  Removing the headphones after this leaves the speakers in the intended, functional state.

### How can the issue be reproduced? (If applicable)

- Boot up the device with no headphones connected.  
- Test a sound-producing event (click the menu or load a game for example)
- No audio output can be detected from he speakers
- Plug in wired (3.5") headphones,  headphones will start playing audio okay.
- Unplug headphones again and speakers are now operational

### What device are you using?
RG353V

### What version of ArkOS are you on?
ArkOS 2.0 (08112023)

### Anything else you'd like to include that may help to review this issue or feature request?
It's not a critical bug but it seems like maybe a (minor) accidental coding error.  Should be considered low priority but is probably an easy fix

More than happy to provide more context, or to test potential fixes.  If you could point me in roughly the right direction I would probably enjoy attempting resolution myself if you could give me a lead?

christianhaitian, 2023-08-27:
Already known. Limit of the current kernel. See here for the workaround. https://github.com/christianhaitian/arkos/issues/731

closed 2023-08-27 by christianhaitian

fexofenadine, 2023-08-27:
brilliant thank you mate