christianhaitian/arkos #779: Charging Problem RGB10S

rendyzou89, 2023-08-29:
Update to ArkOS v2.0 (08/11/2023)

Power Off device > Plung the cable = charge screen wakeup then sleep after 10 sec, then it won't charge (green light)
Power On >  plung the cable on Sleep Mode = didn't charge (Green Light)
Power On > plung the cable on Screen Saver = Charged
Power On > plung on the cable on Playing = Charged

christianhaitian, 2023-08-29:
rk3326 devices won't charge while sleeping.  They should charge while on or off though.  If it's not charging while off, that's not something with the OS, however, if it's showing a green light while plugged in, that means the device thinks the battery is fully charged already.

rendyzou89, 2023-08-30:
Got it, thanks bro

> rk3326 devices won't charge while sleeping. They should charge while on or off though. If it's not charging while off, that's not something with the OS, however, if it's showing a green light while plugged in, that means the device thinks the battery is fully charged already.

rendyzou89, 2023-08-30:
Case close

closed 2023-08-30 by rendyzou89