christianhaitian/arkos #782: Hello, which version compatible with the Handheld R35S?

PONTES1971, 2023-09-07:
### Before submitting your issue or feature request.  DO NOT REQUEST ASSISTANTS FOR PRELOADED IMAGES.  I do not support or condone the use of preloaded ArkOS images.  Preloaded meaning images that come with copyrighted roms preincluded.  Look for support from the creator or supplier of those preloaded images, not from here!  If a ticket is submitted for assistance for such images or if there's an indiciation that a preloaded image is what the support request is for, the issue ticket will be closed with no further response.

1. ***This is not the Stock OS that comes with your device, AmberElec, JelOS, or UnOfficialOS.***  ArkOS is a different distribution from those systems and will have differing features.  The only somewhat similarity is that they both use EmulationStation as a frontend.  Even with that, the versions of EmulationStation that's used in between ArkOS and these other distributions is different with different capabilities.  If there are features you prefer with the other available distributions, just use those distributions instead.  The choice is yours.

2. Have you reviewed the Frequently Asked Questions section available for your supported device in the wiki?

3. If your issue is related to being able to run certain emulators or ports, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

4. Have you reviewed the ***known issues*** section available at https://github.com/christianhaitian/arkos/wiki/Known-issues ?

### Describe the Issue (If applicable)
A clear and concise description of what the bug is.

### How can the issue be reproduced? (If applicable)

### What device are you using?

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen

### Anything else you'd like to include that may help to review this issue or feature request?

christianhaitian, 2023-09-07:
I believe that unit is a clone of the rgb20s which is a clone of the rg351mp.  Good luck.

closed 2023-09-07 by christianhaitian

PONTES1971, 2023-09-07:
That's what I thought, I tried but I couldn't. I will try other versions. Any suggestion? Really thanks in advance

christianhaitian, 2023-09-07:
What are you getting after you flash the image?

PONTES1971, 2023-09-07:
There is a white screen disproportionate to the display.
[image]

christianhaitian, 2023-09-07:
That may mean the screen is not the same as that of the rg351mp I support.  Try copying the `.dtb` file from BOOT partition of the stock os sd card to the BOOT partition of the rg351mp ArkOS image you flashed flashed and rename it to `rk3326-rg351mp-linux.dtb` replacing the existing one there.  See if that works.

MasterREAVer86, 2023-09-07:
I too have the same problem. My R35S is a few hours old, and the cheap sd it came with already died.. I can't copy anythin off of it, and I made a new SDcard with your RG351MP rom, but the I also have the same screen problem. If you, or anyone has the original file that christianhaitian mentioned, please send it to me (if it works for you). Thank you!

christianhaitian, 2023-09-07:
A community member from the AmberElec discord was able to provide this dtb file from the stock os.  It's within the attached zip file.  Extract and try what I suggested above with the RG351MP image.

[rk3326-r35s-linux.zip](https://github.com/christianhaitian/arkos/files/12553184/rk3326-r35s-linux.zip)

MasterREAVer86, 2023-09-07:
thanks christianhaitian,but it is the same as before :(

christianhaitian, 2023-09-07:
Yea the person on the AmberElec discord just shared he tried this already.  Trying something else in the meantime.

christianhaitian, 2023-09-07:
This one seems to work for the r35s per limited testing from a person on the AmberElec discord.
[rk3326-rg351mp-linux.zip](https://github.com/christianhaitian/arkos/files/12553858/rk3326-rg351mp-linux.zip)

MasterREAVer86, 2023-09-08:
YES! The last one works, installed with a beutiful screen, and tried one game that ran nicely! You are a real hero christianhaitian!! Thank you so very much!

christianhaitian, 2023-09-08:
Cool. Enjoy

PONTES1971, 2023-09-08:
How cool Christian! Thanks! I am going to try! I hope I can do it. Anyway, I'll leave here a zip of the original R35S Boot para testarem, com o original não consegui.

[BOOT R35S.zip](https://github.com/christianhaitian/arkos/files/12563228/BOOT.R35S.zip)

PONTES1971, 2023-09-09:
I also got it with the Amberlec boot! Thanks so much Christian!

christianhaitian, 2023-09-09:
It was reported that volume buttons don’t working using AmberElec and this file.

MasterREAVer86, 2023-09-09:
It works for just fine

colemancda, 2023-10-01:
Just wanted to report success with ArkOS_RG351MP_v2.0_09292023.img and the file provided here.

Buxaxa, 2024-06-20:
> HI Christian. Im new here and also new to trying to make this R36s work. Could you shed any light on this..... I did all the things recommended on your page. the ArkOS_RG351MP_v2.0_02232024 ( and i even tried the other version just to be sure) it flashes fine onto the SD card. when i install it i get a double reboot like im told. BUT then the screen is Black. Its backlit (which means its on but just black) and i can here the clicking of the dpad but no screen.
> 
> any advice?

Did you solved your problem? I have the exact same thing... Thanks

birdybird123, 2024-06-20:
I have the same problem used all the panel files
Still black screen