christianhaitian/arkos #785: WIKI: Add additional Linux Instructions

mindovermiles262, 2023-09-11:
Hi, Thanks for this great product.

Linux users can use the `dd` command to successfully copy the ArkOS.img onto their SD cards using the following:

After extracting the image, grab the mount point for your SD card. In my case it's the 14.4G `sda` mount:
```
$ lsblk
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
sda           8:0    1  14.4G  0 disk 
nvme1n1     259:0    0   477G  0 disk 
├─[REDACTED]
```

Then, use `dd` to copy the contents of the ArkOS Image onto your SD Card:

```
$ sudo dd -if=/path/to/ArkOS.img of=/dev/sda [-bs=4M]
```

I just tested this on Kubuntu 20.04, ArkOS V2 05-04-203. I'd love to add this to the wiki but have no permissions to do so. If you could add this I would certainly appreciate it. Thank you!

christianhaitian, 2023-09-11:
Awesome.  Thank you.  This has been added.

closed 2023-09-11 by christianhaitian

Rasenberry, 2023-09-13:
It took me too long to add this and  @mindovermiles262 beat me. I wanted to confirm that this is the best/ easiest method across Linux systems. 

I can confirm the following worked from a steamdeck running in desktop mode with no additional software.

`sudo dd if=/home/deck/Downloads/ArkOS_RG351V_v2.0_01152023.img of=/dev/sda bs=1M status=progress`

WARNING: Just make sure you have the Out-File correct(of=) as it will wipe the disk and it won't always be sda. There are many ways to figure out which one is the SD card(lsblk, fdisk -l, Partition Manager, etc...)

vajra108, 2024-05-11:
Hello all! It can be even easier, you don´t even need to decompress first. 
`xz -dc ArkOS_IMAGE.img.xz | sudo dd of=/dev/sdX status=progress`

mdeguzis, 2024-05-19:
Please correct the example and clarify the text in the brackets `[...]` is optional. It is `if=`, NOT `-if=`

clankfish, 2024-07-04:
> Please correct the example and clarify the text in the brackets `[...]` is optional. It is `if=`, NOT `-if=`. Same goes for `-bs=` being `bs=`.
> 
> My example:
> 
> ```
> sudo dd if=<ARK_OS>.img of=/dev/<DISK_LOCATION> status=progress
> ```
> 
> Use `lsblk` to get the disk location, or the disks utility.

yep, thank you. this confused me as a noob :))

mindovermiles262, 2024-07-04:
Thanks everyone for the catch on the typo and formatting. Not sure why I typed it like that, but the initial post has been updated. Hope that makes this more clear