christianhaitian/arkos #786: Add .m3u support for PICO-8 for multi-carts

BREAKHEART1, 2023-09-13:
Example with Another world demake (https://freds72.itch.io/another-world):

Extract all of the game files into a directory, for example /roms/pico8/aw/
You'll have multiple .p8 carts from the .zip file.
The main cart is aw.p8.

Within the pico8 rom folder, create a m3u file - /roms/pico8/Another_world.m3u
The m3u file should point to the location of the main cart aw/aw.p8.

Load this Another_world.m3u file from ES to launch the game.

christianhaitian, 2023-09-13:
Which pico-8 emulator supports .m3u files?  I don't see evidence that fake-08 or the official raspberry pi pico-8 binaries support .m3u files.

BREAKHEART1, 2023-09-13:
Sorry you are correct, its my fault as I was getting confused by something I saw on Google.

Sorry to waste your time!

closed 2023-09-13 by BREAKHEART1