christianhaitian/arkos #787: Scrap manuel/documentation

kamelie1706, 2023-09-13:
### Before submitting your issue or feature request.  DO NOT REQUEST ASSISTANTS FOR PRELOADED IMAGES.  I do not support or condone the use of preloaded ArkOS images.  Preloaded meaning images that come with copyrighted roms preincluded.  Look for support from the creator or supplier of those preloaded images, not from here!  If a ticket is submitted for assistance for such images or if there's an indiciation that a preloaded image is what the support request is for, the issue ticket will be closed with no further response.

1. ***This is not the Stock OS that comes with your device, AmberElec, JelOS, or UnOfficialOS.***  ArkOS is a different distribution from those systems and will have differing features.  The only somewhat similarity is that they both use EmulationStation as a frontend.  Even with that, the versions of EmulationStation that's used in between ArkOS and these other distributions is different with different capabilities.  If there are features you prefer with the other available distributions, just use those distributions instead.  The choice is yours.

2. Have you reviewed the Frequently Asked Questions section available for your supported device in the wiki?

3. If your issue is related to being able to run certain emulators or ports, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

4. Have you reviewed the ***known issues*** section available at https://github.com/christianhaitian/arkos/wiki/Known-issues ?

### Describe the Issue (If applicable)
Add possibility to scrap/read documentation/manual (pdf)

### How can the issue be reproduced? (If applicable)
yes

### What device are you using?
Anbernic RG353PS

### What version of ArkOS are you on?
2.0

### Anything else you'd like to include that may help to review this issue or feature request?

christianhaitian, 2023-09-13:
Hi, This is not a feature I'm interested in adding support for at this time.

closed 2023-09-13 by christianhaitian

kamelie1706, 2023-10-07:
if at some points you are interested there is this emulation station plugin 
https://github.com/dale-personal/retropie-game-manuals